meta-debian/meta-debian #2: build of meta-debian fails

jeremiah, 2015-04-07:
Hi,

I'm following the instructions from the slides you shared at ELC. I'm getting an error when I build;

Exception: ExpansionError: Failure expanding variable SRCPV, expression was ${@bb.fetch2.get_srcrev(d)} which triggered exception MalformedUrl: The URL: '${DEBIAN_GIT_URI}/debian-xutils-dev
.git;protocol=${DEBIAN_GIT_PROTOCOL};branch=daisy-master' is invalid and cannot be interpreted

The above text is the end of a long python stack trace. The whole stack trace is here: http://pastebin.com/ReC7s3JF

ystk, 2015-04-07:
Thanks for testing meta-debian. This is the first time for us to see this error.

Please follow the README to setup the meta-debian layer.
The slide has a bug to setup meta-debian layer. It was missing information to copy conf/local.conf file by the following command:
  # cp ../poky/meta-debian/conf/local.conf.sample conf/local.conf

Then, please check the conf/bblayers.conf file. It shoud not have meta-yocto as the follows.
--------------------------------------------------
BBLAYERS = " \
  /path/to/poky/meta \
  /path/to/poky/meta-debian \
  "
BBLAYERS_NON_REMOVABLE = " \
  /path/to/poky/meta \
  "
--------------------------------------------------

jeremiah, 2015-04-08:
Thank you, I will try and fix my conf/bblayers.conf file.

closed 2015-04-08 by ystk