meta-debian/meta-debian #36: core-image-minimal fails with IMAGE_FEATURES += "package-management"

arjunneupane, 2016-01-30:
I tried to exten core-image-minimal with "package-managment" but failed, seems apt version from meta-debian is not compatible with poky jethro.

DEBUG: SITE files ['endian-little', 'bit-32', 'arm-common', 'arm-32', 'common-linux', 'common-glibc', 'arm-linux', 'arm-linux-gnueabi', 'common']
DEBUG: Executing shell function do_compile
NOTE: make -j 4
Compiling aptconfiguration.cc to /home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/build/obj/apt-pkg/aptconfiguration.opic
Compiling indexrecords.cc to /home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/build/obj/apt-pkg/indexrecords.opic
Compiling sourcelist.cc to /home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/build/obj/apt-pkg/sourcelist.opic
Compiling deb/debrecords.cc to /home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/build/obj/apt-pkg/debrecords.opic
Compiling deb/debversion.cc to /home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/build/obj/apt-pkg/debversion.opic
Compiling cacheset.cc to /home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/build/obj/apt-pkg/cacheset.opic
Compiling deb/debindexfile.cc to /home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/build/obj/apt-pkg/debindexfile.opic
Compiling deb/debsrcrecords.cc to /home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/build/obj/apt-pkg/debsrcrecords.opic
/home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/apt-1.0.10.1/apt-pkg/aptconfiguration.cc: In static member function 'static const std::vector<std::basic_string<char> > APT::Configuration::getLanguages(const bool&, const bool&, const char*$
/home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/apt-1.0.10.1/apt-pkg/aptconfiguration.cc:196:30: error: expected primary-expression before '(' token
  string const envMsg = string(Locale == 0 ? std::setlocale(LC_MESSAGES, NULL) : *Locale);
                              ^
/home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/apt-1.0.10.1/apt-pkg/aptconfiguration.cc:196:79: error: expected unqualified-id before ':' token
  string const envMsg = string(Locale == 0 ? std::setlocale(LC_MESSAGES, NULL) : *Locale);
                                                                               ^
../buildlib/library.mak:65: recipe for target '/home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/build/obj/apt-pkg/aptconfiguration.opic' failed
make[2]: *** [/home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/build/obj/apt-pkg/aptconfiguration.opic] Error 1
make[2]: *** Waiting for unfinished jobs....
Makefile:14: recipe for target 'all' failed
make[1]: *** [all] Error 2
makefile:18: recipe for target 'all' failed
make: *** [all] Error 2
ERROR: oe_runmake failed
ERROR: Function failed: do_compile (log file is located at /home/ane/yocto-test/build/tmp/work/armv5e-debian-linux-gnueabi/apt/1.0.10.1-r0/temp/log.do_compile.26622)

Build Configuration:
BB_VERSION        = "1.28.0"
BUILD_SYS         = "x86_64-linux"
NATIVELSBSTRING   = "Debian-8.3"
TARGET_SYS        = "arm-debian-linux-gnueabi"
MACHINE           = "qemuarm"
DISTRO            = "debian"
DISTRO_VERSION    = "8.0"
TUNE_FEATURES     = "arm armv5 thumb dsp"
TARGET_FPU        = "soft"
meta              = "jethro:2fb7ee2628e23d7efc9b041bb9daae7c4a8de541"
meta-debian       = "jethro:8965b11a96cc483971b64dfd2dd21d6b59d3

zuka0828, 2016-02-08:
Thank you for reporting the issue.

The error occurs while compiling
	meta/recipes-devtools/apt/apt_1.0.10.1.bb
This recipe is not in meta-debian but OE-Core, so the cause may come from some conflicts between OE-Core apt and cross toolchain of meta-debian.

Currently, "package-management" feature is not available in meta-debian. In order to enable "package-management" feature, apt_debian.bb needs to be implemented in meta-debian (Currently, only "apt-native" is available). We might support this feature in future, but the priority of supporting the feature is not so high because of some reasons:

1. When users install/upgrade packages in their rootfs with apt, anyway they need to bitbake packages again, and then update apt repository by themselves. This takes as much time as to bitbake rootfs again and re-install.

2. meta-debian doesn't support installing foreign packages from other distros' apt repositories. Installing foreign packages is not recommended because the binaries in deb packages of other distros and their metadata (package name, dependencies, etc.)are not compatible with packages in rootfs generated with meta-debian. Some foreign packages might be able to be installed into rootfs, but we cannot assure that they work correctly with regular packages in the rootfs.

However, installing/upgrading packages in rootfs with apt is sometimes useful for debugging purposes (ex: temporally install -dbg packages for debugging and remove them). Therefore, we will support at least dpkg in rootfs. If apt is strongly needed for your system, we'd like to consider adding apt_debian.bb.

arjunneupane, 2016-02-20:
Thank you very much for taking time to reply and for considering to add apt_debian.bb.
And for sure, it would be nice to have feature.

zuka0828, 2017-05-15:
`IMAGE_FEATURES += "package-management"` is already available in meta-debian jethro branch.
apt commands in target rootfs are also usable for installing/uninstalling packages.
Also, `${DEPLOY_DIR_DIR}` in your build directory can be used as an apt repository for your rootfs by exporting the directory through web server or just copying the directory to target rootfs.

closed 2017-05-15 by zuka0828