meta-debian/meta-debian #68: Please bump vulnerable Expat to 2.2.1

hartwork, 2017-06-20:
Please excuse my posting style.

Vulnerable copy found at:
https://github.com/meta-debian/meta-debian/blob/master/recipes-debian/expat/expat_debian.bb

Expat 2.2.1 change log:
https://github.com/libexpat/libexpat/blob/master/expat/Changes

Commits that you may need on top:
* https://github.com/libexpat/libexpat/commit/602e6c78ca750c082b72f8cdf4a38839b312959f
* https://github.com/libexpat/libexpat/commit/e5e25fac1698d9ea8dbb2e51c18d1072254b5f4b

Best, Sebastian

ystk, 2017-07-03:
Thank you very much for the notification.

Instead of bump the version from 2.1.0-6 to 2.2.1, Deby imported security fixes from Debian project.
Expat_debian.bb recipe refers to the source code at the following URL.
    - https://github.com/ystk/debian-expat/tree/jessie-master
The current version of above tree is 2.1.0-6+deb8u4 which includes security fixes for the following CVEs.
    - CVE-2017-9233, external entity infinite loop bug,
    - CVE-2016-9063, undefined behavior from signed integer overflow.

hartwork, 2017-07-03:
I'm not too happy about bumps to old-stable but it's your decision. if you want to stick with old-stable, this ticket can be closed as far as I am concerned.

ystk, 2017-07-10:
I close this issue.

closed 2017-07-10 by ystk