meta-debian/meta-debian #74: Unable to determine endianness for architecture 'INVALID' ERROR: Please add your architecture to siteinfo.bbclass

naveen230, 2017-10-10:
I'm trying to use this layer for rpi3 devices

I changed the MACHINE="raspberrypi3" in local.conf and added layers in bblayers.conf

But I'm getting this error.I used the "morty" branch instead of Jethro.Is that an issue?

tuyenhv, 2017-10-11:
Hi!

Could you please try to set MACHINE = "raspberrypi-v1" ?
It should be consistent with meta-debian/bsp/**meta-raspberrypi-v1**.conf.

Regards,

tuyenhv, 2017-10-11:
Hi!

Could you please try to set MACHINE = "raspberrypi-v1" ?
It should be consistent with meta-debian/bsp/meta-raspberrypi/conf/machine/**raspberrypi-v1**.conf.

Regards,

naveen230, 2017-10-12:
Thanks for your reply tuyenhv
Now I'm getting this error ,
 make[4]: Leaving directory '/root/build/tmp/work/x86_64-linux/gettext-native/0.19.3-0/build/gettext-tools/doc'
| Makefile:1684: recipe for target 'all-recursive' failed
| make[3]: *** [all-recursive] Error 1
| ERROR: oe_runmake failed
| make[3]: Leaving directory '/root/build/tmp/work/x86_64-linux/gettext-native/0.19.3-0/build/gettext-tools'
| Makefile:1549: recipe for target 'all' failed
| make[2]: *** [all] Error 2
| make[2]: Leaving directory '/root/build/tmp/work/x86_64-linux/gettext-native/0.19.3-0/build/gettext-tools'
| Makefile:368: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/root/build/tmp/work/x86_64-linux/gettext-native/0.19.3-0/build'
| Makefile:323: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /root/build/tmp/work/x86_64-linux/gettext-native/0.19.3-0/temp/log.do_compile.10457)
ERROR: Task (virtual:native:/root/poky/meta-debian/recipes-debian/gettext/gettext_debian.bb:do_compile) failed with exit code '1'
I tried
bitbake -c clean gettext-native
bitbake gettext-native
but getting same error when I run 2nd command

tuyenhv, 2017-10-13:
Hi!
Could you provide the GCC version in your Host machine?
We usually use Debian Jessie as Host machine (GCC version: 4.9.2) 
and don't get the error.

naveen230, 2017-10-16:
It was 5.x version before, now I downgraded to 4.9.4 and still getting the error.I'm trying to use this layer raspberrypi3.Will it work?

naveen230, 2017-10-16:
Downgrading gcc and perl works but leads to a new issue with dpkg
Do you think we should add architecture to dpkg manually?

the error is,
WARNING: arm not found in cputable
| checking dpkg operating system type... linux-gnueabi
| configure: WARNING: linux-gnueabi not found in ostable
| checking dpkg architecture name... configure: error: cannot determine host dpkg architecture
| WARNING: exit code 1 from a shell command.

more detailed error from log file,

checking whether arm-deby-linux-gnueabi-g++ -march=armv7ve -marm -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a7 --sysroot=/root/build/tmp/sysroots/raspberrypi3 -std=gnu++11 accepts -Wformat-security... no checking whether arm-deby-linux-gnueabi-g++ -march=armv7ve -marm -mfpu=neon-vfpv4 -mfloat-abi=hard -mcpu=cortex-a7 --sysroot=/root/build/tmp/sysroots/raspberrypi3 -std=gnu++11 accepts -Wpointer-arith.
--

sangorrin, 2019-02-15:
Closed since there is no more activity and the solution (using Debian Jessie) as host is already given.

closed 2019-02-15 by sangorrin