meta-debian/meta-debian #76: [Questian] Building Deby for Beaglebone Black

elfgoh, 2017-10-19:
Apologies for cross posting. But I thought it may be wise to post here as well, especially there is nice markdown support

Iread on this [presentation](http://events.linuxfoundation.org/sites/events/files/slides/ISAR-DEBY-OSSJ2017_r10.pdf) that Deby supports Beaglebone Black. I tried with the steps listed at the git repo README and change machine in 
`build/conf/local.conf` to: `MACHINE ?= "beaglebone"`

However, I get the error
```
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    Please set a valid MACHINE in your local.conf or environment
```

Please advise me how I can safely resolve this error. Thanks!

https://groups.google.com/forum/#!topic/meta-debian/FY_eQ3KiOr0

naveen230, 2017-10-19:
navigate to build/conf
touch sanity.conf

elfgoh, 2017-10-19:
Thanks for the suggestion @naveen230 This is the resulting output

```
luther@debian:~/meta-debian/build$ cd conf/
luther@debian:~/meta-debian/build/conf$ touch sanity.conf
luther@debian:~/meta-debian/build/conf$ cd ..
luther@debian:~/meta-debian/build$ time bitbake core-image-minimal
ERROR: /home/luther/meta-debian/poky/meta-debian/recipes-misc/ltp/ltp_git.bb: Unable to determine endianness for architecture 'INVALID'
ERROR: /home/luther/meta-debian/poky/meta-debian/recipes-misc/ltp/ltp_git.bb: Please add your architecture to siteinfo.bbclass
ERROR: Failed to parse recipe: /home/luther/meta-debian/poky/meta-debian/recipes-misc/ltp/ltp_git.bb
ERROR: /home/luther/meta-debian/poky/meta-debian/recipes-kernel/linux/linux-base_git.bb: cannot map 'INVALID' to a linux kernel architecture

Summary: There were 4 ERROR messages shown, returning a non-zero exit code.
```

naveen230, 2017-10-19:
Try adding meta-blackboard layer in bblayers.conf
https://github.com/beagleboard/meta-beagleboard

elfgoh, 2017-10-19:
I have cloned [meta-beagleboard](https://github.com/beagleboard/meta-beagleboard) in `build/conf`

For `bblayers.conf`, I have edited it to be as follows:

```
# LAYER_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
LCONF_VERSION = "6"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/luther/meta-debian/poky/meta \
  /home/luther/meta-debian/poky/meta-debian \
  /home/luther/meta-debian/poky/meta-beagleboard \
  "
BBLAYERS_NON_REMOVABLE ?= " \
  /home/luther/meta-debian/poky/meta \
  /home/luther/meta-debian/poky/meta-debian \
  /home/luther/meta-debian/poky/meta-beagleboard \
  "
```

But this results in the following error: 

```
luther@debian:~/meta-debian/build$ time bitbake core-image-minimal
ERROR: Traceback (most recent call last):
  File "/home/luther/meta-debian/poky/bitbake/lib/bb/cookerdata.py", line 181, in wrapped
    return func(fn, *args)
  File "/home/luther/meta-debian/poky/bitbake/lib/bb/cookerdata.py", line 206, in parse_config_file
    return bb.parse.handle(fn, data, include)
  File "/home/luther/meta-debian/poky/bitbake/lib/bb/parse/__init__.py", line 113, in handle
    return h['handle'](fn, data, include)
  File "/home/luther/meta-debian/poky/bitbake/lib/bb/parse/parse_py/ConfHandler.py", line 126, in handle
    abs_fn = resolve_file(fn, data)
  File "/home/luther/meta-debian/poky/bitbake/lib/bb/parse/__init__.py", line 136, in resolve_file
    raise IOError(errno.ENOENT, "file %s not found" % fn)
FileNotFoundError: [Errno 2] file /home/luther/meta-debian/poky/meta-beagleboard/conf/layer.conf not found

ERROR: Unable to parse /home/luther/meta-debian/poky/meta-beagleboard/conf/layer.conf: [Errno 2] file /home/luther/meta-debian/poky/meta-beagleboard/conf/layer.conf not found
```

naveen230, 2017-10-19:
goto meta-beagleboard/conf/
touch layer.conf

elfgoh, 2017-10-19:
Thanks!

Still not quite there yet

```
luther@debian:~/meta-debian/build$ time bitbake core-image-minimal
ERROR: /home/luther/meta-debian/poky/meta-debian/recipes-misc/ltp/ltp_git.bb: Unable to determine endianness for architecture 'INVALID'
ERROR: /home/luther/meta-debian/poky/meta-debian/recipes-misc/ltp/ltp_git.bb: Please add your architecture to siteinfo.bbclass
ERROR: Failed to parse recipe: /home/luther/meta-debian/poky/meta-debian/recipes-misc/ltp/ltp_git.bb
ERROR: /home/luther/meta-debian/poky/meta-debian/recipes-kernel/linux/linux-base_git.bb: cannot map 'INVALID' to a linux kernel architecture
ERROR: /home/luther/meta-debian/poky/meta-debian/recipes-kernel/linux-libc-headers/linux-libc-headers-base_git.bb: cannot map 'INVALID' to a linux kernel architecture

Summary: There were 5 ERROR messages shown, returning a non-zero exit code.

```

naveen230, 2017-10-20:
I'm building the same for raspberrypi3 so I have to add other dependencies like meta-openembedded as well.So I think you have to do the same.

elfgoh, 2017-10-20:
I assume it is this repo? https://github.com/openembedded/meta-openembedded.git

elfgoh, 2017-10-20:
Are you documenting your progress for raspberry pi 3? Would love to try it on my pi, and also use it for a reference for beaglebone black

naveen230, 2017-10-25:
Yes I'm documenting.Getting lot of package errors.Will let you know once I'm successful

vsivanagulu, 2018-04-24:
are you completed any documents for this meta-debin.

sangorrin, 2019-02-15:
Please check this:
https://gitlab.com/cip-project/cip-core/deby/tree/master/poky/meta-cip-bbb

closed 2019-02-15 by sangorrin

vsivanagulu, 2019-02-15:
I am folling the above link, But when running below connamd Its giving error
 
`debian-yocto$ docker run -v $PWD/deby:/deby -e USER_ID=`id -u $USER` -e http_proxy=$http_proxy -e https_proxy=$https_proxy -e NO_PROXY="$no_proxy" -it kasproject/kas:0.13.0 sh
`

> docker run -v $PWD/deby:/deby -e USER_ID=`id -u $USER` -e http_proxy=$http_proxy -e https_proxy=$https_proxy -e NO_PROXY="$no_proxy" -it kasproject/kas:0.13.0 sh
> docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.38/containers/create: dial unix /var/run/docker.sock: connect: permission denied.
> See 'docker run --help'.

sangorrin, 2019-02-15:
https://docs.docker.com/install/linux/linux-postinstall/

sangorrin, 2019-02-20:
> I am folling the above link, But when running below connamd Its giving error
> 
> `debian-yocto$ docker run -v $PWD/deby:/deby -e USER_ID=`id -u $USER`-e http_proxy=$http_proxy -e https_proxy=$https_proxy -e NO_PROXY="$no_proxy" -it kasproject/kas:0.13.0 sh`
> NOTE: I am using ubuntu based system. should i need to use debian based os?
> 
> > docker run -v $PWD/deby:/deby -e USER_ID=`id -u $USER` -e http_proxy=$http_proxy -e https_proxy=$https_proxy -e NO_PROXY="$no_proxy" -it kasproject/kas:0.13.0 sh
> > docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.38/containers/create: dial unix /var/run/docker.sock: connect: permission denied.
> > See 'docker run --help'.

Were you successful with $ sudo usermod -aG docker $USER?