meta-debian/meta-debian #106: Building debian for cl-imx7d

amol1000, 2018-10-03:
Hello, 
 I am trying to build a custom rootfs that relies heavily on debian packages. When i try to incorporate meta-debian in my yocto build, I get the following error:
  
ERROR: /home/amol/test/fsl-arm-yocto-bsp/sources/meta-debian/recipes-misc/qemu/qemu-bios-native_1.7.0.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 2, function: <module>
     0001:__anon_25__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_conf_machine_include_arm_feature_arm_thumb_inc(d)
 *** 0002:__anon_647__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_classes_base_bbclass(d)
     0003:__anon_60__home_amol_test_fsl_arm_yocto_bsp_sources_meta_fsl_arm_classes_fsl_dynamic_packagearch_bbclass(d)
     0004:__anon_1319__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_classes_insane_bbclass(d)
     0005:__anon_242__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_classes_package_bbclass(d)
     0006:__anon_747__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_classes_package_rpm_bbclass(d)
File: '/home/amol/test/fsl-arm-yocto-bsp/sources/poky/meta/classes/base.bbclass', lineno: 552, function: __anon_647__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_classes_base_bbclass
     0548:    needsrcrev = False
     0549:    srcuri = d.getVar('SRC_URI', True)
     0550:    for uri in srcuri.split():
     0551:        (scheme, _ , path) = bb.fetch.decodeurl(uri)[:3]
 *** 0552:
     0553:        # HTTP/FTP use the wget fetcher
     0554:        if scheme in ("http", "https", "ftp"):
     0555:            d.appendVarFlag('do_fetch', 'depends', ' wget-native:do_populate_sysroot')
     0556:
File: '/home/amol/test/fsl-arm-yocto-bsp/sources/poky/bitbake/lib/bb/fetch2/__init__.py', lineno: 365, function: decodeurl
     0361:    """
     0362:
     0363:    m = re.compile('(?P<type>[^:]*)://((?P<user>[^/]+)@)?(?P<location>[^;]+)(;(?P<parm>.*))?').match(url)
     0364:    if not m:
 *** 0365:        raise MalformedUrl(url)
     0366:
     0367:    type = m.group('type')
     0368:    location = m.group('location')
     0369:    if not location:
Exception: MalformedUrl: The URL: '${MISC_GIT_URI}/${MISC_GIT_PREFIX}qemu.git;protocol=${MISC_GIT_PROTOCOL};branch=stable-2.1' is invalid and cannot be interpreted

ERROR: Failed to parse recipe: /home/amol/test/fsl-arm-yocto-bsp/sources/meta-debian/recipes-misc/qemu/qemu-bios-native_1.7.0.bb
ERROR: /home/amol/test/fsl-arm-yocto-bsp/sources/meta-debian/recipes-misc/ltp/ltp_git.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 2, function: <module>
     0001:__anon_25__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_conf_machine_include_arm_feature_arm_thumb_inc(d)
 *** 0002:__anon_647__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_classes_base_bbclass(d)
     0003:__anon_60__home_amol_test_fsl_arm_yocto_bsp_sources_meta_fsl_arm_classes_fsl_dynamic_packagearch_bbclass(d)
     0004:__anon_1319__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_classes_insane_bbclass(d)
     0005:__anon_242__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_classes_package_bbclass(d)
     0006:__anon_747__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_classes_package_rpm_bbclass(d)
File: '/home/amol/test/fsl-arm-yocto-bsp/sources/poky/meta/classes/base.bbclass', lineno: 552, function: __anon_647__home_amol_test_fsl_arm_yocto_bsp_sources_poky_meta_classes_base_bbclass
     0548:    needsrcrev = False
     0549:    srcuri = d.getVar('SRC_URI', True)
     0550:    for uri in srcuri.split():
     0551:        (scheme, _ , path) = bb.fetch.decodeurl(uri)[:3]
 *** 0552:
     0553:        # HTTP/FTP use the wget fetcher
     0554:        if scheme in ("http", "https", "ftp"):
     0555:            d.appendVarFlag('do_fetch', 'depends', ' wget-native:do_populate_sysroot')
     0556:
File: '/home/amol/test/fsl-arm-yocto-bsp/sources/poky/bitbake/lib/bb/fetch2/__init__.py', lineno: 365, function: decodeurl
     0361:    """
     0362:
     0363:    m = re.compile('(?P<type>[^:]*)://((?P<user>[^/]+)@)?(?P<location>[^;]+)(;(?P<parm>.*))?').match(url)
     0364:    if not m:
 *** 0365:        raise MalformedUrl(url)
     0366:
     0367:    type = m.group('type')
     0368:    location = m.group('location')
     0369:    if not location:
Exception: MalformedUrl: The URL: '${MISC_GIT_URI}/${MISC_GIT_PREFIX}ltp.git;protocol=${MISC_GIT_PROTOCOL}' is invalid and cannot be interpreted

sangorrin, 2019-02-15:
meta-debian v1 (morty) is not supposed to be combined with layers created for yocto which use different source code and policies. You can give a try to the master-ng branch which is in development and tries to achieve more compatibility with yocto.

closed 2019-02-15 by sangorrin

zuka0828, 2019-02-16:
Hi @amol1000 
Please use `master` branch instead of `master-ng` which is an old name of `master` and will be removed soon.