meta-debian/meta-debian #134: util-linux lsblk lacks libsmartcols runtime support

sangorrin, 2019-03-22:
Reported by Patryk Mungai from Renesas: https://lava.ciplatform.org/scheduler/job/896#L890

lsblk: /lib/arm-linux-gnueabihf/libsmartcols.so.1: version `SMARTCOLS_2.30' not found (required by lsblk)
lsblk: /lib/arm-linux-gnueabihf/libsmartcols.so.1: version `SMARTCOLS_2.27' not found (required by lsblk)
lsblk: /lib/arm-linux-gnueabihf/libsmartcols.so.1: version `SMARTCOLS_2.30' not found (required by lsblk)
lsblk: /lib/arm-linux-gnueabihf/libsmartcols.so.1: version `SMARTCOLS_2.27' not found (required by lsblk)
lsblk: /lib/arm-linux-gnueabihf/libsmartcols.so.1: version `SMARTCOLS_2.30' not found (required by lsblk)
lsblk: /lib/arm-linux-gnueabihf/libsmartcols.so.1: version `SMARTCOLS_2.27' not found (required by lsblk)

dothanhtrung, 2019-03-22:
Hi Daniel san,

In the log file, the test copies `libsmartcols.so.1` and `lsblk` from an `additional_lib_bin`. Are these files provided by meta-debian:morty.

https://lava.ciplatform.org/scheduler/job/896#L885
```sh
cp -r ./additional_lib_bin/lib/libblkid.so.1 ./additional_lib_bin/lib/libmount.so.1 ./additional_lib_bin/lib/libsmartcols.so.1 ./additional_lib_bin/lib/libuuid.so.1 /lib/
cp -r ./additional_lib_bin/usr_bin/lsblk /usr/bin/
```

Currently, meta-debian:morty provides util-linux 2.25 but the error is about finding 2.27/2.30.
Also, in the log error, `lsblk` tries to find `libsmartcols.so.1` in `/lib/arm-linux-gnueabihf`, but the above code copies `libsmartcols.so.1` to `/lib` instead.

patrykmungai, 2019-03-25:
Hello @dothanhtrung ,
Thank you for spotting the difference. I can confirm that copying the libraries to /lib/arm-linux-gnueabihf solved the issue.

patrykmungai, 2019-03-26:
Update: Removing all the static binaries and libraries that were prebuilt for the test results in the test working as well. There is nothing wrong with lsblk that comes with Deby. Apologies for my mistake.
This issue can be closed.

zuka0828, 2019-03-26:
@patrykmungai Thank you for the confirmation. I would like to close this issue.

closed 2019-03-26 by zuka0828