meta-debian/meta-debian #143: running debian source is required. gcc-source-8.3.0-8.3.0-r0 do_fetch fails. Needs upgrade to gcc-8_8.3.0-6

PieterGit, 2019-04-21:
I'm trying to build meta-debian with an Intel Edison. See 
https://github.com/htot/meta-intel-edison/pull/2 

It fails with downloading
http://ftp.debian.org/debian/pool/main/g/gcc-8/gcc-8_8.3.0-5.dsc;name=gcc-8_8.3.0-5.dsc
Current gcc version is http://ftp.debian.org/debian/pool/main/g/gcc-8/gcc-8_8.3.0-6.dsc

Details:
```
pietergit@ubuntu1810:~/warrior/out/linux64/build$ bitbake core-image-minimal
WARNING: Host distribution "ubuntu-18.10" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Parsing recipes: 100% |##########################################| Time: 0:02:26
Parsing of 2274 .bb files complete (0 cached, 2274 parsed). 3414 targets, 347 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-18.10"
TARGET_SYS           = "i686-deby-linux"
MACHINE              = "edison"
DISTRO               = "deby"
DISTRO_VERSION       = "10.0"
TUNE_FEATURES        = "m32 corei7"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "warrior:0e392026ffefee098a890c39bc3ca1f697bacb52"
meta-oe              
meta-python          
meta-networking      = "master:d42cce95a8717c3e5bba3701b71f2f9822c2d00c"
meta-intel           = "master:ed44056d2c8ecbbb5ca75f1e490c9dc7daa876e3"
meta-intel-edison-bsp 
meta-intel-edison-distro = "warrior:b6bd2603652549c347cec4ce7a56439fb98539a1"
meta-acpi            = "eds-5.0.0:6a1fb60a23f3f03887771b53b9ad529385a19f5b"
meta-debian          = "master:91c198bb38bbd31e9ae8ad8b2067e3b11c845913"

Initialising tasks: 100% |#######################################| Time: 0:00:10
Sstate summary: Wanted 438 Found 0 Missed 438 Current 0 (0% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: gcc-source-8.3.0-8.3.0-r0 do_fetch: Failed to fetch URL http://ftp.debian.org/debian/pool/main/g/gcc-8/gcc-8_8.3.0-5.dsc;name=gcc-8_8.3.0-5.dsc, attempting MIRRORS if available
ERROR: gcc-source-8.3.0-8.3.0-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="3603"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/home/pietergit/warrior/out/linux64/poky/scripts:/home/pietergit/warrior/out/linux64/build/tmp/work-shared/gcc-8.3.0-r0/recipe-sysroot-native/usr/bin/i686-deby-linux:/home/pietergit/warrior/out/linux64/build/tmp/work-shared/gcc-8.3.0-r0/recipe-sysroot/usr/bin/crossscripts:/home/pietergit/warrior/out/linux64/build/tmp/work-shared/gcc-8.3.0-r0/recipe-sysroot-native/usr/sbin:/home/pietergit/warrior/out/linux64/build/tmp/work-shared/gcc-8.3.0-r0/recipe-sysroot-native/usr/bin:/home/pietergit/warrior/out/linux64/build/tmp/work-shared/gcc-8.3.0-r0/recipe-sysroot-native/sbin:/home/pietergit/warrior/out/linux64/build/tmp/work-shared/gcc-8.3.0-r0/recipe-sysroot-native/bin:/home/pietergit/warrior/out/linux64/poky/bitbake/bin:/home/pietergit/warrior/out/linux64/build/tmp/hosttools"; export HOME="/home/pietergit"; /usr/bin/env wget -t 2 -T 30 --passive-ftp --no-check-certificate -P /home/pietergit/warrior/out/linux64/build/downloads 'http://ftp.debian.org/debian/pool/main/g/gcc-8/gcc-8_8.3.0-5.dsc' --progress=dot -v failed with exit code 8, output:
--2019-04-20 14:22:46--  http://ftp.debian.org/debian/pool/main/g/gcc-8/gcc-8_8.3.0-5.dsc
Resolving ftp.debian.org (ftp.debian.org)... 130.89.148.12, 2001:67c:2564:a119::148:12
Connecting to ftp.debian.org (ftp.debian.org)|130.89.148.12|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2019-04-20 14:22:46 ERROR 404: Not Found.

ERROR: gcc-source-8.3.0-8.3.0-r0 do_fetch: Fetcher failure for URL: 'http://ftp.debian.org/debian/pool/main/g/gcc-8/gcc-8_8.3.0-5.dsc;name=gcc-8_8.3.0-5.dsc'. Unable to fetch URL from any source.
ERROR: gcc-source-8.3.0-8.3.0-r0 do_fetch: 
ERROR: gcc-source-8.3.0-8.3.0-r0 do_fetch: Function failed: base_do_fetch
ERROR: Logfile of failure stored in: /home/pietergit/warrior/out/linux64/build/tmp/work-shared/gcc-8.3.0-r0/temp/log.do_fetch.28228
ERROR: Task (/home/pietergit/warrior/meta-debian/recipes-debian/gcc/gcc-source_debian.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 63 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/pietergit/warrior/meta-debian/recipes-debian/gcc/gcc-source_debian.bb:do_fetch
Summary: There were 2 WARNING messages shown.
Summary: There were 4 ERROR messages shown, returning a non-zero exit code.
```

I managed to get my build fixed by using the comment from:
https://github.com/meta-debian/meta-debian/pull/129

I changed `DEBIAN_SOURCE_ENABLED` to `"1"` in `build-dir/conf/local.conf`. It now fetches the gcc files.

iwamatsu, 2019-04-27:
Hi,

This is because gcc-8_8.3.0-5.dsc has been removed from the Debian repository.
If you do not want to update the software version or you do not want to use DEBIAN_SOURCE_ENABLED, you can work around this issue by specifying "0" for DEBIAN_USE_SNAPSHOT.

dothanhtrung, 2019-05-02:
Hi PieterGit,

Thank you for report the problem and sorry for late reply.

> It fails with downloading
> http://ftp.debian.org/debian/pool/main/g/gcc-8/gcc-8_8.3.0-5.dsc;name=gcc-8_8.3.0-5.dsc
> Current gcc version is http://ftp.debian.org/debian/pool/main/g/gcc-8/gcc-8_8.3.0-6.dsc

That's my bad for not keep updating the gcc version.
Now, gcc version has been updated to 8.3.0-6 since commit 51c6c0a.
You may want to use branch "warrior" instead of "master".

zuka0828, 2019-05-10:
@PieterGit Thank you for your reporting.
The problem has been fixed by 51c6c0a as @dothanhtrung said, so I would like to close this issue.
Please re-open this discussion if the problem is still there.

closed 2019-05-10 by zuka0828

PieterGit, 2019-05-11:
@zuka0828 @dothanhtrung Thanks for fixing. I'll switch to the `warrior` branch.