meta-debian/meta-debian #164: Failure while compiling apt - Yocto

1BE6-ankit, 2019-07-31:
I am testing meta-debian on a Ubuntu 18.04 virtual machine. The only major thing that I have added is the following lines in my build/conf/local.conf file:
```
IMAGE_FEATURES += " package-management"

IMAGE_INSTALL_append = " apt"
```

But, apt fails to compile with the following error:

```
ERROR: apt-1.2.24-r0 do_compile: oe_runmake failed
ERROR: apt-1.2.24-r0 do_compile: Function failed: do_compile (log file is located at /home/preempt/debian-preempt/poky/build/tmp/work/core2-64-deby-linux/apt/1.2.24-r0/temp/log.do_compile.110040)
ERROR: Logfile of failure stored in: /home/preempt/debian-preempt/poky/build/tmp/work/core2-64-deby-linux/apt/1.2.24-r0/temp/log.do_compile.110040
Log data follows:
| DEBUG: SITE files ['endian-little', 'common-linux', 'common-glibc', 'bit-64', 'x86_64-linux', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 1
| Generating POT file /home/preempt/debian-preempt/poky/build/tmp/work/core2-64-deby-linux/apt/1.2.24-r0/build/po/apt.pot
| mv: cannot stat '/home/preempt/debian-preempt/poky/build/tmp/work/core2-64-deby-linux/apt/1.2.24-r0/build/po/domains/apt/c.pot.tmp': No such file or directory
| ERROR: oe_runmake failed
| makefile:32: recipe for target '/home/preempt/debian-preempt/poky/build/tmp/work/core2-64-deby-linux/apt/1.2.24-r0/build/po/apt.pot' failed
| make[2]: *** [/home/preempt/debian-preempt/poky/build/tmp/work/core2-64-deby-linux/apt/1.2.24-r0/build/po/apt.pot] Error 1
| Makefile:14: recipe for target 'all' failed
| make[1]: *** [all] Error 2
| makefile:18: recipe for target 'all' failed
| make: *** [all] Error 2
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/preempt/debian-preempt/poky/build/tmp/work/core2-64-deby-linux/apt/1.2.24-r0/temp/log.do_compile.110040)
ERROR: Task (/home/preempt/debian-preempt/poky/meta/recipes-devtools/apt/apt_1.2.24.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 987 tasks of which 986 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/preempt/debian-preempt/poky/meta/recipes-devtools/apt/apt_1.2.24.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

dothanhtrung, 2019-08-01:
Hi,

It seems that apt is not supported in 'poky-tiny' (which 'deby-tiny' bases on). You can pass this problem by changing DISTRO from 'deby-tiny' to 'deby' in your builddir/conf/local.conf or adding `USE_NLS = "yes"` into recipe apt at poky/meta/recipes-devtools/apt/apt_1.2.24.bb.

zuka0828, 2019-08-08:
@1BE6-ankit I confirmed that the same error occurs without meta-debian when we specify `DISTRO = "poky-tiny"`, so this is not a meta-debian specific issue.

We are no plan to fix this at the moment because `poky-tiny` (and `deby-tiny`) does not assume that `apt` is installed in the target rootfs.

Please consider using `DISTRO = deby` (as @dothanhtrung said) if it can match your usecase.

1BE6-ankit, 2019-08-11:
@dothanhtrung @zuka0828 though late to reply, thanks for your help.

closed 2019-08-11 by 1BE6-ankit