meta-debian/meta-debian #182: glib-2.0: failed to build nativesdk

cheat-sc, 2019-11-20:
```
| Checking if "pthread_setname_np(const char*)" with dependency threads links: NO
| Checking if "pthread_setname_np(pthread_t, const char*)" with dependency threads links: NO
| WARNING: Cross property growing_stack is using default value False
| Checking for function "iconv_open" : NO
| 
| meson.build:1702:4: ERROR:  Problem encountered: iconv implementation "libc" not found
```
|layer|branch|rev|
|-----|-------|---|
|poky|warrior|c9a1a608f5146e5dcfebc8e566cb7f6388a3986f|
|meta-debian|warrior|f3af4d4c6de5295fed93b108d7efb000aa623ac4|

yuwata, 2019-11-20:
This seems caused by Poky's f61f37bef1eb234e1de85f10da0056250950704f.

yuwata, 2019-11-20:
This can be fixed by https://github.com/yuwata/meta-debian/commit/b7e9f0024fb61013410cfe95fe6f1b5fb3d4094d. But I believe that this issue should be fixed in Poky side.

cheat-sc, 2019-11-20:
I have confirmed this problem does not occur when I used poky (c9a1a608f5146e5dcfebc8e566cb7f6388a3986f) only.

yuwata, 2019-11-20:
Thanks. So, my guess was wrong, as this issue only occurs when meta-debian built with the problematic poky commit. I will submit the commit as a PR.

yuwata, 2019-11-26:
Note that my build configuration is the following:
```
Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-deby-linux"
MACHINE              = "qemuarm64"
DISTRO               = "deby-tiny"
DISTRO_VERSION       = "10.0"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 
meta-poky            = "warrior:c9a1a608f5146e5dcfebc8e566cb7f6388a3986f"
meta-debian          = "warrior-tmp:69ed92c580067b1e757b15fbb18a87b3ea70b9bc"
```
Here, warrior-tmp branch is provided by #185.

zuka0828, 2020-01-08:
I believe that this issue has been fixed by #197, closed.

closed 2020-01-08 by zuka0828

cheat-sc, 2020-01-22:
Thanks for everything :)