meta-debian/meta-debian #191: Yocto Zeus support

reneherrero, 2019-12-08:
Yocto Zeus has been release and now includes Buster in the supported distributions.

Any plans to update this project to Zeus?

zuka0828, 2019-12-17:
@reneherrero Thank you for your question.

Currently, we have no plan to support Zeus or newer Yocto releases for Debian 10 sources and will focus on Warrior.

The reasons why we selected Warrior as our target is the recipes in this release (branch) are closer to the package versions supported in Debian 10, which means the compatibilities and maintainability between meta-debian recipes and Yocto Project recipes are better than other releases.

Supporting Buster as the build environment in Zeus is also good news, but it may not be motivative enough to change our target because meta-debian(warrior) is mainly tested on Buster already and there are risks in the compatibility and maintainability from the perspective above.

reneherrero, 2019-12-17:
Fair enough, thanks for the response.

closed 2019-12-17 by reneherrero

heysion, 2021-06-09:
Hi , @zuka0828 Iam interested in porting  meta-debian to support the yocto LTS version - Warrior , but I don't know what workload is involved. Could you give me a roadmap and advice help me  try it ?

hiraku-wfs, 2021-06-09:
Hi @heysion, If you are interested in LTS for more packages rather than build system, there are extra packages in [meta-debian-extended](https://github.com/miraclelinux/meta-debian-extended).

This can be used together with meta-debian. You can use deby(-tiny) as DISTRO. (I usually use these layers with [meta-emlinux](https://github.com/miraclelinux/meta-emlinux)).

meta-debian and meta-debian-extended cover core-image-weston image except for some non-executable packages such as fonts.

heysion, 2021-06-10:
> Hi @heysion, If you are interested in LTS for more packages rather than build system, there are extra packages in [meta-debian-extended](https://github.com/miraclelinux/meta-debian-extended).
>
> This [...]

Thanks a million. I will try it as soon as for my best. but I was interested for used yocto build a debian buster system.

zuka0828, 2021-06-13:
Hello @heysion , thank you for your question.

> I was interested for used yocto build a debian buster system.

The thing you want to do is to make the recipes in meta-debian warrior branch, which are for building Debian buster source packages, available with poky dunfell branch?

So far, there was no plan to do this activity for dunfell (LTS release) because there is not so many benefits in our situation.
The things we need to do for the porting would be

* Try building meta-debian layer with poky dunfell
* Find and fix the issues (in meta-debian recipes) that come from compatibility problems between the recipes in poky meta-debian is expecting (warrior based) and dunfell's
    * Implemented features in `.inc` in poky that meta-debian recipes depend on. This would be bigger if more gaps between package versions of buster and dunfell.
    * Implemented features in common resources in poky like class files, configs, bitbake, etc.If specifications in dunfell are updated from warrior and they are not compatible with ones meta-debian recipes are expecting, some efforts to fix them are required.
* Continuously maintain meta-debian recipes following future updates in poky dunfell
    * More fixes would be required if some functional changes affecting compatibility with meta-debian happen in poky dunfell. I think there are few concerns about this point because Yocto Project stable/LTS mainly allows security fixes and bug fixes only [0].

We may only know the actual workload for the porting above after try building and testing all meta-debian recipes with dunfell.

My understanding is that the value of this porting is a balance of required efforts for fixing compatibility issues v.s. gain from newer recipes in dunfell (e.g. additional useful features in .bbclass files) + longer term maintenance for them in Yocto Project. If you are using less poky own recipes that are not provided in not in meta-debian, less benefits to be based on dunfell.

It's worth starting discussions for the porting if there are strong demands and positive for contributions.

[0] https://wiki.yoctoproject.org/wiki/Stable_Release_and_LTS#Stable.2FLTS_Patch_Acceptance_Policies

heysion, 2021-06-15:
Hello @zuka0828  , Thanks a lot of  your reply.
> The thing you want to do is to make the recipes in meta-debian warrior branch, which are for building Debian buster source packages, available with poky dunfell branch?

Actually , I wanted build debian in the poky dunfell branch which I was fixed some software version and copy many files to directory meta and meta-debian . But i can't get debian image on my pay out . So far I try  poky in the warrior branch but didn't get debian image . Now I try to fix meta-debian on the poky warrior branch . 
`poky commit c8c383d958807b991e31d22f612ba2a81a3860a4 (HEAD -> warrior, origin/warrior))`

If i can holed on and keep doing it , I think try to the poky dunfell branch  porting meta-debian . The dunfell branch many package versionswas high for debian  e.g. toolchain , i don't have solution for it .Maybe need rewrite and override some recipe  .

zuka0828, 2021-07-07:
Hi @heysion ,

> Actually , I wanted build debian in the poky dunfell branch which I was fixed some software version and copy many files to directory meta and meta-debian .

I still don't fully understand what you want to have.
Like the followings or something else?

```
$ git clone -b dunfell git://git.yoctoproject.org/poky.git
$ git clone -b dunfell https://github.com/meta-debian/meta-debian.git poky/meta-debian  # No dunfell branch actually
$ export TEMPLATECONF=meta-debian/conf
$ source ./poky/oe-init-build-env
$ bitbake core-image-minimal
```

> Maybe need rewrite and override some recipe .

Yes. I quickly tried `bitbake core-image-minimal` with poky dunfell + meta-debian warrior then found at least more than 10 recipes including python need to be significantly updated. For example, some `.bbclass` and `.inc` files that meta-debian recipes `inherit` or `require` are missing, which means we need to rewrite some parts of these recipes from scratch or find other replacements then test them. I think that they are very small parts of what to be solve and more recipes, classes, configs, etc. would need to be updated.

Also, I'm interested in your motivation to use dunfell. Is it longer-term maintenance by the community? You already have 
your own recipes working with dunfell and want to make them available with meta-debian? etc.