meta-debian/meta-debian #216: Build failed in some recipes.

watatuki, 2020-02-12:
I built core-image-minimal as described in README.md today.

Some recipe failed by download error.

Failed recipes:
  virtual:native:/cross/yocto/debian/poky/meta-debian/recipes-debian/nss/nss_debian.bb:do_fetch
  virtual:native:/cross/yocto/debian/poky/meta-debian/recipes-debian/e2fsprogs/e2fsprogs_debian.bb:do_fetch
  virtual:native:/cross/yocto/debian/poky/meta-debian/recipes-debian/python/python3_debian.bb:do_fetch
  /cross/yocto/debian/poky/meta-debian/recipes-debian/base-files/base-files_debian.bb:do_fetch
  /cross/yocto/debian/poky/meta-debian/recipes-debian/qemu/qemu-native_debian.bb:do_fetch
  /cross/yocto/debian/poky/meta-debian/recipes-debian/qemu/qemu-system-native_debian.bb:do_fetch

Log example:
ERROR: e2fsprogs-native-1.44.5-r0 do_fetch: Fetcher failure: Fetch command export 
--2020-02-12 14:22:44--  http://ftp.debian.org/debian/pool/main/e/e2fsprogs/e2fsprogs_1.44.5-1+deb10u2.dsc
Resolving ftp.debian.org (ftp.debian.org)... 2001:67c:2564:a119::148:12, 130.89.148.12
Connecting to ftp.debian.org (ftp.debian.org)|2001:67c:2564:a119::148:12|:80... failed: No route to host.
Connecting to ftp.debian.org (ftp.debian.org)|130.89.148.12|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2020-02-12 14:23:02 ERROR 404: Not Found.

ERROR: e2fsprogs-native-1.44.5-r0 do_fetch: Fetcher failure for URL: 'http://ftp.debian.org/debian/pool/main/e/e2fsprogs/e2fsprogs_1.44.5-1+deb10u2.dsc;name=e2fsprogs_1.44.5-1+deb10u2.dsc'. Unable to fetch URL from any source.
ERROR: e2fsprogs-native-1.44.5-r0 do_fetch:
ERROR: e2fsprogs-native-1.44.5-r0 do_fetch: Function failed: base_do_fetch

I checked to ftp site. 
http://ftp.debian.org/debian/pool/main/e/e2fsprogs/e2fsprogs_1.44.5-1+deb10u2.dsc is not available.
http://ftp.debian.org/debian/pool/main/e/e2fsprogs/e2fsprogs_1.44.5-1+deb10u3.dsc is available.

My question: how to fix this case ?

One of the solution:
modify to meta-debian/recipes-debian/sources/e2fsprogs.inc

Is this correct?

zuka0828, 2020-02-13:
@watatuki Thank you for your report.

I believe that https://github.com/meta-debian/meta-debian/commit/972c612a3adc0bea2d39cf76921815e3db33edaa fixes this issue.
The cause is that several old Debian source packages are removed from Debian apt repositories (after 10.3 released I guess). In such cases, we need to update corresponding recipes to the latest source versions.

watatuki, 2020-02-17:
Thank you response and bug fix.
I succeeded in all fetch task of core-image-minimal.

closed 2020-02-17 by watatuki