meta-debian/meta-debian #220: Should aide be built with static link option?

yuwata, 2020-03-16:
Currently, `aide` is built with shared libraries:
```
# ldd /usr/bin/aide
	linux-vdso.so.1 (0x0000ffffac384000)
	libm.so.6 => /lib/libm.so.6 (0x0000ffffac253000)
	libpcre.so.3 => /usr/lib/libpcre.so.3 (0x0000ffffac205000)
	libmhash.so.2 => /usr/lib/libmhash.so.2 (0x0000ffffac1c9000)
	libacl.so.1 => /usr/lib/libacl.so.1 (0x0000ffffac1b0000)
	libselinux.so.1 => /lib/libselinux.so.1 (0x0000ffffac177000)
	libaudit.so.1 => /lib/libaudit.so.1 (0x0000ffffac140000)
	libe2p.so.2 => /lib/libe2p.so.2 (0x0000ffffac127000)
	libz.so.1 => /lib/libz.so.1 (0x0000ffffac100000)
	libc.so.6 => /lib/libc.so.6 (0x0000ffffabf95000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffffac356000)
	libattr.so.1 => /usr/lib/libattr.so.1 (0x0000ffffabf7f000)
	libdl.so.2 => /lib/libdl.so.2 (0x0000ffffabf6b000)
	libcap-ng.so.0 => /lib/libcap-ng.so.0 (0x0000ffffabf55000)
```
But, Debian seems to build `aide` with static linked libraries (and provides aide-shared package for shared linked version). What is the policy of meta-debian about the shared- or static-linked programs?

In general, `aide` checks the file modification, so it is better to be built statistically to reduce its dependencies. But when read-only root filesystem e.g., squashfs, is used, it is not so important to check the `aide` itself and its dependent libraries, then dynamic-link seems ok in that case. What is the primal usecase for aide in meta-debian?

sangorrin, 2020-10-05:
@yuwata san thanks for your feedback.
You can now use the following to enable static linking for `aide`
https://gitlab.com/cip-project/cip-core/deby/-/blob/cip-core-buster/meta-cip-security/conf/include/aide-static-libs.inc

Please note that in order to build aide statically, aide dependencies need to be built statically as well.

Thanks,
Daniel

masami256, 2020-10-07:
@sangorrin Thank you for your advice.
We will use aide-static-libs.inc to create static linked binary.

@yuwata  It's okay to close this issue.  thanks.

yuwata, 2020-10-07:
@sangorrin Thank you for your support!

@masami256 Sure.

closed 2020-10-07 by yuwata