meta-debian/meta-debian #250: creating recipe for bro package

venkata-pyla, 2020-12-01:
I have created this issue here to get some help from community and resolve the issue facing bro recipe.

The initial changes for creating bro recipe are pushed in this PR #249 

Below is the error I am getting which points out to the 'bifcl' command not found which was compiled as part of bro build and uses it in the later stages of the bro build.
```
ERROR: bro-2.5.5-r0 do_compile: Function failed: do_compile (log file is located at /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/temp/log.do_compile.136)
ERROR: Logfile of failure stored in: /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/temp/log.do_compile.136
Log data follows:
| DEBUG: SITE files ['endian-little', 'common-linux', 'common-glibc', 'bit-64', 'x86_64-linux', 'common']
| DEBUG: Executing shell function do_compile
|      /usr/include     /usr/lib     /lib     /lib     /usr/share  /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/build/src
| NOTE: VERBOSE=1 cmake --build /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/build --target all -- -j 4
| [1/785] [BIFCL] Processing const.bif
| FAILED: src/const.bif.func_def src/const.bif.func_h src/const.bif.func_init src/const.bif.netvar_def src/const.bif.netvar_h src/const.bif.netvar_init scripts/base/bif/const.bif.bro
| cd /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/build/src && bifcl /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/bro-2.5.5-minimal/src/const.bif || ( rm -f /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/build/src/const.bif.func_def /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/build/src/const.bif.func_h /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/build/src/const.bif.func_init /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/build/src/const.bif.netvar_def /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/build/src/const.bif.netvar_h /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/build/src/const.bif.netvar_init && exit 1 ) && /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/recipe-sysroot-native/usr/bin/cmake -E copy const.bif.bro /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/build/scripts/base/bif/const.bif.bro && /work/build/tmp/work/core2-64-deby-linux/bro/2.5.5-r0/recipe-sysroot-native/usr/bin/cmake -E remove -f const.bif.bro
| /bin/sh: 1: bifcl: not found
```
I am not sure how to specify the 'bifcl' path which was built already and present in the path
 or the 'bifcl' binary should be build as native first and then depends on the native package in bro recipe

venkata-pyla, 2020-12-01:
> Another way to know about bro is to check existing metadata in other distros.
> e.g.
> * Debian (I’m not sure if bro package is already support cross-building or not, but it’s worth to check debian/rules once)
> * buildroot (Not an official recipe?):  https://github.com/alexisfacques/buildroot-bro/blob/master/package/bro/bro.mk
> 
> If you cannot find any solution to build bifcl for host without bro-native, please implement bro-native to provide that.

as suggested by @zuka0828 i am currently investigating the above points.

zuka0828, 2021-01-31:
@venkata-pyla Thanks for your work to add bro and its dependencies.

Please check the following updates:

* I have merged PR #249 anyway because it includes enough bro metadata which should be the base of further patches
* `bifcl-native` recipe added (95ad1320e7281deb3a669fc7abdbbfd00f5b3728)
* Clean up `DEPENDS` of bro recipe (8701fd926958d4873ce249c02273e1425f931bb0)

Now, `do_configure` and `do_compile` of bro recipe finishes successfully.
The following QA issue is only remaining. 

```
ERROR: bro-2.5.5-r0 do_package_qa: QA Issue: bro: /work/armv7vet2hf-neon-deby-linux-gnueabi/bro/2.5.5-r0/packages-split/bro/usr/bin/bro contains probably-redundant RPATH /usr/lib [useless-rpaths]
ERROR: bro-2.5.5-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
ERROR: bro-2.5.5-r0 do_package_qa: 
ERROR: bro-2.5.5-r0 do_package_qa: Function failed: do_package_qa
```

Please confirm that the bro recipe can be built successfully in your environment, then take a look at the QA issue.

venkata-pyla, 2021-02-01:
@zuka0828 thank for providing the bifcl-native and resolving the dependency issues.

I will check the QA issues

venkata-pyla, 2021-02-01:
@zuka0828 san, fixed QA issue in this PR https://github.com/meta-debian/meta-debian/pull/255, 
kindly review and provide your feedback

Thanks

zuka0828, 2021-02-01:
@venkata-pyla Thank you for the patch, I confirmed that now `bro` recipe can be built correctly, so would like to close this issue.

FYI: I added a minor update for `bro` recipe to change `site` directory. 5d9b36e4121f71939bed74babf0af449e5618b50

closed 2021-02-01 by zuka0828

venkata-pyla, 2021-02-01:
Thank you @zuka0828 san for considering the patch and changing site directory