meta-debian/meta-debian #256: [morty] python-cryptograph: do_compile() error

zuka0828, 2021-02-25:
Build conditions

* Host: Debian 8 amd64
* Target: `MACHINE = qemux86`
* Command: `bitbake world`

log.do_compile:

```
DEBUG: Executing shell function do_compile
ERROR: python setup.py build execution failed.
running build
running build_ext
building '_Cryptography_cffi_fc665d23x4f158fee' extension
creating /home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/git/cryptography/hazmat/primitives/__pycache__/cryptography
creating /home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/git/cryptography/hazmat/primitives/__pycache__/cryptography/hazmat
creating /home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/git/cryptography/hazmat/primitives/__pycache__/cryptography/hazmat/primitives
creating /home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/git/cryptography/hazmat/primitives/__pycache__/cryptography/hazmat/primitives/__pycache__
i586-deby-linux-gcc -m32 -march=i586 --sysroot=/home/deby/poky/build-mibs/tmp/sysroots/qemux86 -DNDEBUG -g -O3 -Wall -Wstrict-prototypes -fno-strict-aliasing -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0=/usr/src/debug/python-cryptography/0.6.1-r0 -fdebug-prefix-map=/home/deby/poky/build-mibs/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/deby/poky/build-mibs/tmp/sysroots/qemux86= -fPIC -I/home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/python2.7 -c cryptography/hazmat/primitives/__pycache__/_Cryptography_cffi_fc665d23x4f158fee.c -o /home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/git/cryptography/hazmat/primitives/__pycache__/cryptography/hazmat/primitives/__pycache__/_Cryptography_cffi_fc665d23x4f158fee.o
i586-deby-linux-gcc -m32 -march=i586 --sysroot=/home/deby/poky/build-mibs/tmp/sysroots/qemux86 -shared -Wl,-O1 -Wl,--hash-style=gnu -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0=/usr/src/debug/python-cryptography/0.6.1-r0 -fdebug-prefix-map=/home/deby/poky/build-mibs/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/deby/poky/build-mibs/tmp/sysroots/qemux86= /home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/git/cryptography/hazmat/primitives/__pycache__/cryptography/hazmat/primitives/__pycache__/_Cryptography_cffi_fc665d23x4f158fee.o -o /home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/git/cryptography/hazmat/primitives/__pycache__/_Cryptography_cffi_fc665d23x4f158fee.so
Traceback (most recent call last):
  File "setup.py", line 338, in <module>
    **keywords_with_side_effects(sys.argv)
  File "/home/deby/poky/build-mibs/tmp/sysroots/x86_64-linux/usr/lib/python2.7/distutils/core.py", line 151, in setup
    dist.run_commands()
  File "/home/deby/poky/build-mibs/tmp/sysroots/x86_64-linux/usr/lib/python2.7/distutils/dist.py", line 953, in run_commands
    self.run_command(cmd)
  File "/home/deby/poky/build-mibs/tmp/sysroots/x86_64-linux/usr/lib/python2.7/distutils/dist.py", line 971, in run_command
    cmd_obj.ensure_finalized()
  File "/home/deby/poky/build-mibs/tmp/sysroots/x86_64-linux/usr/lib/python2.7/distutils/cmd.py", line 109, in ensure_finalized
    self.finalize_options()
  File "setup.py", line 106, in finalize_options
    self.distribution.ext_modules = get_ext_modules()
  File "setup.py", line 83, in get_ext_modules
    from cryptography.hazmat.primitives import constant_time, padding
  File "/home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/git/cryptography/hazmat/primitives/constant_time.py", line 37, in <module>
    ext_package="cryptography",
  File "/home/deby/poky/build-mibs/tmp/sysroots/x86_64-linux/usr/lib/python2.7/dist-packages/cffi/api.py", line 340, in verify
    lib = self.verifier.load_library()
  File "/home/deby/poky/build-mibs/tmp/sysroots/x86_64-linux/usr/lib/python2.7/dist-packages/cffi/verifier.py", line 75, in load_library
    return self._load_library()
  File "/home/deby/poky/build-mibs/tmp/sysroots/x86_64-linux/usr/lib/python2.7/dist-packages/cffi/verifier.py", line 151, in _load_library
    return self._vengine.load_library()
  File "/home/deby/poky/build-mibs/tmp/sysroots/x86_64-linux/usr/lib/python2.7/dist-packages/cffi/vengine_cpy.py", line 149, in load_library
    raise ffiplatform.VerificationError(error)
cffi.ffiplatform.VerificationError: importing '/home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/git/cryptography/hazmat/primitives/__pycache__/_Cryptography_cffi_fc665d23x4f158fee.x86_64-linux-gnu.so': /home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/git/cryptography/hazmat/primitives/__pycache__/_Cryptography_cffi_fc665d23x4f158fee.x86_64-linux-gnu.so: wrong ELF class: ELFCLASS32
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_compile (log file is located at /home/deby/poky/build-mibs/tmp/work/i586-deby-linux/python-cryptography/0.6.1-r0/temp/log.do_compile.3333958)
```

hieunv-tsdv, 2021-03-09:
Dear @zuka0828 san,

I created a patch to fix this issue: (https://github.com/meta-debian/meta-debian/pull/273)

Python-native can't import module from ".so" library with target architecture.

zuka0828, 2021-03-09:
Fixed by 1b192003dc92bdf093564d0774819b076e3349c1
Thanks @hieunv-tsdv san !

PR #273 (2021-03-09): python-cryptography: correct build dependency
Use hazmat module from python-cryptography-native
to avoid importing error:
| ImportError: wrong ELF class: ELFCLASS32

Fix #256