meta-debian/meta-debian #257: pulseaudio: do_debian_patch() failed

zuka0828, 2021-02-25:
Build conditions: Same as #256 

log.do_debian_patch:

```
DEBUG: SITE files ['endian-little', 'bit-32', 'ix86-common', 'common-linux', 'common-glibc', 'i586-linux', 'common']
DEBUG: Executing shell function do_debian_patch
NOTE: Debian source format is '3.0 (quilt)'
Applying patch 0001-raop-Don-t-use-incompatible-default-sample-spec-para.patch
patching file src/modules/raop/module-raop-sink.c

Applying patch 0002-Use-the-fixed-point-speex-resampler-on-ARM.patch
patching file man/pulse-daemon.conf.5.xml.in
patching file src/daemon/daemon.conf.in
patching file src/pulsecore/core.c
patching file src/pulsecore/resampler.c

Applying patch 0003-exit-with-X-session.patch
patching file src/daemon/start-pulseaudio-x11.in
patching file src/daemon/start-pulseaudio-kde.in

Applying patch 0004-dont-start-when-already-running.patch
patching file src/daemon/start-pulseaudio-kde.in
patching file src/daemon/start-pulseaudio-x11.in

Applying patch kfreebsd_no_lock_and_threads_synchro.patch
patching file src/daemon/main.c

Applying patch only-autostart-kde-version.patch
patching file src/daemon/pulseaudio.desktop.in

Applying patch rtp-recv-fix-crash-on-empty-UDP-packets-CVE-2014-397.patch
patching file src/modules/rtp/rtp.c

Applying patch zsh-completion-Fix-completion-for-default-sink-sourc.patch
patching file shell-completion/pulseaudio-zsh-completion.zsh

Applying patch gnu-kfreebsd.patch
patching file src/daemon/caps.c
patching file src/pulsecore/atomic.h
patching file src/pulsecore/sconv_sse.c
patching file src/pulsecore/svolume_mmx.c
patching file src/pulsecore/svolume_sse.c
patching file src/tests/once-test.c
patching file src/tests/rtstutter.c

Applying patch mix-Fix-mixing-of-S24-samples-stored-as-S32RE.patch
patching file src/pulsecore/mix.c
patching file src/tests/mix-test.c

Applying patch tests-Cleanup-mix-test.patch
patching file src/tests/mix-test.c

Applying patch endianmacros-Fix-PA_MAYBE_INT16_SWAP-macro.patch
patching file src/pulsecore/endianmacros.h

Applying patch tests-Fix-mix-test-on-big-endian-systems.patch
patching file src/tests/mix-test.c

Applying patch util-Fix-pa_get_binary_name-on-Debian-kFreeBSD.patch
patching file src/pulse/util.c

Applying patch misc-cleanups-and-bug-fixes.patch
can't find file to patch at input line 12
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|Author: Peter Meerwald <pmeerw@pmeerw.net>
|Description: misc cleanups and bug fixes
| This is a collapse of a few small patches by Peter Meerwald.
| Changes are:
| put *.log and *.trs (which are created by running 'make check') into src/.gitignore
| fix format specifiers, avoid compilation warnings
| fix compilation warning when certain syscalls are not available (e.g. on kFreeBSD)
| add checks for the expected results in sigbus-test
| add SIGSEGV to the memtrap signal handler to make the sigbus-test work on kFreeBSD
|--- a/src/.gitignore
|+++ b/src/.gitignore
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
patching file src/pulsecore/core-util.c
patching file src/pulsecore/memtrap.c
patching file src/tests/mcalign-test.c
patching file src/tests/sigbus-test.c
Patch misc-cleanups-and-bug-fixes.patch does not apply (enforce with -f)
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_debian_patch (log file is located at /home/deby/poky/build-mibs/tmp/work/i586-deby-linux/pulseaudio/5.0-r0/temp/log.do_debian_patch.3756980)
```

hieunv-tsdv, 2021-03-03:
@zuka0828 san, I think we need to update the _pulseaudio_ source code (https://github.com/ystk/debian-pulseaudio.git).

_src/.gitignore_ is renamed to _src/.gitignore-rename_ from _ystk/debian-pulseaudio_.

So we need to update _pulseaudio_ source code to same as Debian jessie, or apply the patch from Okamoto-san to remove  patch for _.gitignore_ file.
[0001-Remove-patch-for-.gitignore.zip](https://github.com/meta-debian/meta-debian/files/6073300/0001-Remove-patch-for-.gitignore.zip)

zuka0828, 2021-03-04:
@hieunv-tsdv Thanks for pointing the cause out.
https://github.com/ystk/debian-pulseaudio/commit/df6e498eb0922281b5469c6129481fd898078a0b is a bad commit which includes both updates from Debian package and our custom changes like adding `*.gitignore-rename` files. Give us time to check the best way to fix this.

zuka0828, 2021-04-02:
I decided to apply the patch above for `jessie-master` branch because this branch should not be modified from the original jessie source package tree.

I believe that the commit https://github.com/meta-debian/meta-debian/commit/2d75fea1c314c10cc03cb29d42bdb98f3c926247 fixes this issue.

closed 2021-04-02 by zuka0828