meta-debian/meta-debian #265: [morty] libsasl2-modules-gssapi-heimdal: do_populate_sysroot() error

zuka0828, 2021-02-25:
Build conditions: Same as #256 

log.do_populate_sysroot:

```
...
ERROR: The recipe libsasl2-modules-gssapi-heimdal is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
   /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/lib/i386-linux-gnu/sasl2/libscram.so.2
 Matched in b'manifest-qemux86-libsasl2-modules-gssapi-mit.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/lib/i386-linux-gnu/sasl2/libscram.so.2.0.25
 Matched in b'manifest-qemux86-libsasl2-modules-gssapi-mit.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/lib/i386-linux-gnu/sasl2/libscram.so
 Matched in b'manifest-qemux86-libsasl2-modules-gssapi-mit.populate_sysroot'
Please verify which recipe should provide the above files.
The build has stopped as continuing in this scenario WILL break things, if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would however there are several different scenarios which can result in this and we don't know which one this is. It may be you have switched providers of something like virtual/kernel (e.g. from linux-yocto to linux-yocto-dev), in that case you need to execute the clean task for both recipes and it will resolve this error. It may be you changed DISTRO_FEATURES from systemd to udev or vice versa. Cleaning those recipes should again resolve this error however switching DISTRO_FEATURES on an existing build directory is not supported, you should really clean out tmp and rebuild (reusing sstate should be safe). It could be the overlapping files detected are harmless in which case adding them to SSTATE_DUPWHITELIST may be the correct solution. It could also be your build is including two different conflicting versions of things (e.g. bluez 4 and bluez 5 and the correct solution for that would be to resolve the conflict. If in doubt, please ask on the mailing list, sharing the error and filelist above.
ERROR: If the above message is too much, the simpler version is you're advised to wipe out tmp and rebuild (reusing sstate is fine). That will likely fix things in most (but not all) cases.
DEBUG: Python function sstate_task_postfunc finished
ERROR: Function failed: sstate_task_postfunc
```

hieunv-tsdv, 2021-03-09:
@zuka0828 san,

This issue same as #267, "libsasl2-modules-gssapi-heimdal" is conflict with "libsasl2-modules-gssapi-mit" recipe.

zuka0828, 2021-03-09:
@hieunv-tsdv Thanks. Close this issue due to the same reason as https://github.com/meta-debian/meta-debian/issues/268#issuecomment-793889297

closed 2021-03-09 by zuka0828