meta-debian/meta-debian #267: [morty] sudo: do_populate_sysroot() error

zuka0828, 2021-02-25:
Build conditions: Same as #256

log.do_populate_sysroot:

```
...
ERROR: The recipe sudo is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
   /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/sudo_plugin.h
 Matched in b'manifest-qemux86-sudo-ldap.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/lib/sudo/sudo_noexec.so
 Matched in b'manifest-qemux86-sudo-ldap.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/lib/sudo/sudoers.so
 Matched in b'manifest-qemux86-sudo-ldap.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/lib/sudo/group_file.so
 Matched in b'manifest-qemux86-sudo-ldap.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/lib/sudo/system_group.so
 Matched in b'manifest-qemux86-sudo-ldap.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/lib/systemd/system/sudo.service
 Matched in b'manifest-qemux86-sudo-ldap.populate_sysroot'
Please verify which recipe should provide the above files.
The build has stopped as continuing in this scenario WILL break things, if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would however there are several different scenarios which can result in this and we don't know which one this is. It may be you have switched providers of something like virtual/kernel (e.g. from linux-yocto to linux-yocto-dev), in that case you need to execute the clean task for both recipes and it will resolve this error. It may be you changed DISTRO_FEATURES from systemd to udev or vice versa. Cleaning those recipes should again resolve this error however switching DISTRO_FEATURES on an existing build directory is not supported, you should really clean out tmp and rebuild (reusing sstate should be safe). It could be the overlapping files detected are harmless in which case adding them to SSTATE_DUPWHITELIST may be the correct solution. It could also be your build is including two different conflicting versions of things (e.g. bluez 4 and bluez 5 and the correct solution for that would be to resolve the conflict. If in doubt, please ask on the mailing list, sharing the error and filelist above.
ERROR: If the above message is too much, the simpler version is you're advised to wipe out tmp and rebuild (reusing sstate is fine). That will likely fix things in most (but not all) cases.
DEBUG: Python function sstate_task_postfunc finished
ERROR: Function failed: sstate_task_postfunc
```

hieunv-tsdv, 2021-03-03:
Hi @zuka0828 san,

The _sudo_ provide two binary packages: _sudo_ and _sudo-ldap_. Both of them provide the same file.
- https://packages.debian.org/buster/amd64/sudo/filelist
- https://packages.debian.org/buster/amd64/sudo-ldap/filelist

Currently, in meta-debian:morty, _sudo_ is seperate to 2 recipes (sudo_debian.bb and sudo-ldap_debian.bb). They can't build together in the same build tree.

If we want to fix the conflicts, we have to ship all file from _sudo-ldap_ to different path than the _sudo_ package,
Or we can drop sudo_ldap recipe from meta-debian:morty (I don't see any recipes depend on sudo-ldap recipe)

Do you have any comments or suggestions about this problem?

zuka0828, 2021-03-04:
@hieunv-tsdv Thanks for your quick analysis.
Let me check if the `sudo-ldap` recipe is in an dependency chain of our existing projects.