meta-debian/meta-debian #268: [morty] neon-gnutls: do_populate_sysroot() error

zuka0828, 2021-02-25:
Build conditions: Same as #256

log.do_populate_sysroot:

```
...
ERROR: The recipe neon-gnutls is trying to install files into a shared area when those files already exist. Those files and their manifest location are:
   /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_ssl.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_i18n.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_alloc.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_acl.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_auth.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_props.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_xmlreq.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_compress.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_pkcs11.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_dates.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_redirect.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_session.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_xml.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_locks.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_string.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_207.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_utils.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_basic.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_socket.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_request.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_acl3744.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_defs.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_uri.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/include/neon/ne_md5.h
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/lib/i386-linux-gnu/pkgconfig/neon.pc
 Matched in b'manifest-qemux86-neon.populate_sysroot'
 /home/deby/poky/build-mibs/tmp/sysroots/qemux86/usr/bin/crossscripts/neon-config
 Matched in b'manifest-qemux86-neon.populate_sysroot'
Please verify which recipe should provide the above files.
The build has stopped as continuing in this scenario WILL break things, if not now, possibly in the future (we've seen builds fail several months later). If the system knew how to recover from this automatically it would however there are several different scenarios which can result in this and we don't know which one this is. It may be you have switched providers of something like virtual/kernel (e.g. from linux-yocto to linux-yocto-dev), in that case you need to execute the clean task for both recipes and it will resolve this error. It may be you changed DISTRO_FEATURES from systemd to udev or vice versa. Cleaning those recipes should again resolve this error however switching DISTRO_FEATURES on an existing build directory is not supported, you should really clean out tmp and rebuild (reusing sstate should be safe). It could be the overlapping files detected are harmless in which case adding them to SSTATE_DUPWHITELIST may be the correct solution. It could also be your build is including two different conflicting versions of things (e.g. bluez 4 and bluez 5 and the correct solution for that would be to resolve the conflict. If in doubt, please ask on the mailing list, sharing the error and filelist above.
ERROR: If the above message is too much, the simpler version is you're advised to wipe out tmp and rebuild (reusing sstate is fine). That will likely fix things in most (but not all) cases.
DEBUG: Python function sstate_task_postfunc finished
ERROR: Function failed: sstate_task_postfunc
```

hieunv-tsdv, 2021-03-09:
@zuka0828 san,

This issue same as #267, "neon" is conflict with "neon-gnutls" recipe.

zuka0828, 2021-03-09:
@hieunv-tsdv Thanks. I confirmed that `libneon27-dev` and `libneon27-gnutls-dev` also conflict each other.

I think there is no reason to remove either recipe because both implementations are fine.
Also, this issue only happens when both recipes are built at the same time, which should be avoided in most cases.
(i.e. User select either recipe to provide neon feature)

For `bitbake world`, only way to solve such conflicts is to use `EXCLUDE_FROM_WORLD`.
If you find other better ways, please let me know.

closed 2021-03-09 by zuka0828