meta-debian/meta-debian #275: [warrior] openssl: host-user-contaminated in do_package_qa()

zuka0828, 2021-03-10:
Build conditions:

* Commit: 1d746eff6256a16809e47c846dd9514ab43a28fb
* Host: Debian 10 amd64
* Target: `MACHINE = qemuarm`
* Command: `bitbake openssl`

The following WARNING happens in `do_package_qa()`.

```
WARNING: openssl-1.1.1d-r0 do_package_qa: QA Issue: openssl: /openssl-conf/etc/ssl/openssl.cnf is owned by uid 1000, which is the same as the user running bitbake. This may be due to host contamination [host-user-contaminated]
```

hiraku-wfs, 2021-03-11:
@zuka0828 Thank you for pointing out the issue. Terasawa-san and I are investigating the cause.

hieunv-tsdv, 2021-04-01:
@hiraku-wfs san, @zuka0828 san,

In my opinion, we can fix this issue by changing the owner of _/etc/ssl/openssl.cnf_ file
```
@@ -147,6 +147,7 @@ do_install () {
        ln -sf ${@oe.path.relative('${libdir}/ssl-1.1', '${sysconfdir}/ssl/certs')} ${D}${libdir}/ssl-1.1/certs
        ln -sf ${@oe.path.relative('${libdir}/ssl-1.1', '${sysconfdir}/ssl/private')} ${D}${libdir}/ssl-1.1/private
        ln -sf ${@oe.path.relative('${libdir}/ssl-1.1', '${sysconfdir}/ssl/openssl.cnf')} ${D}${libdir}/ssl-1.1/openssl.cnf
+       chown -R root:root ${D}${sysconfdir}/ssl/openssl.cnf
```

hieunv-tsdv, 2021-04-01:
@hiraku-wfs san,

Thanks for the information.
I wonder if _openssl_ from _meta_ layer have faced this problem. Our recipe is based on that recipe.
Using the _install_ and _rm_ is a good solution to solve this issue.

zuka0828, 2021-04-01:
@hiraku-wfs @hieunv-tsdv Thank you for your investigation and comments!

I also think that using `install` command is better.

> I wonder if openssl from meta layer have faced this problem. Our recipe is based on that recipe.

It might be caused by some differences between the source trees.
When I have time, I would check the cause why the owner of `openssl.cnf` is only the generic user.

zuka0828, 2021-04-04:
The warning is fixed by #280.
@hiraku-wfs @masami256 Thank you very much for solving this issue.

closed 2021-04-04 by zuka0828