meta-debian/meta-debian #279: [morty] confuse: do_populate_lic() fails

zuka0828, 2021-04-01:
```
ERROR: confuse-2.7-r0 do_populate_lic: QA Issue: confuse: The LIC_FILES_CHKSUM does not match for file://support/missing;beginline=6;endline=25;md5=40b23661ee5b2d5c1ec9e09e5e928e56
confuse: The new md5 checksum is 0c1612fda3d56544f336bf595f0ed822
confuse: Check if the license information has changed in /home/deby/poky/build-world/tmp/work/i586-deby-linux/confuse/2.7-r0/git/support/missing (lines 6 through to 25) to verify that the LICENSE value "GPLv2+ & GPLv3+" remains valid [license-checksum]
ERROR: confuse-2.7-r0 do_populate_lic: QA Issue: confuse: The LIC_FILES_CHKSUM does not match for file://support/config.guess;beginline=7;endline=25;md5=e443b5b496d01ee52219d5269560d697
confuse: The new md5 checksum is 69c5a9e1f3642cbb28f2922e996ca3eb
confuse: Check if the license information has changed in /home/deby/poky/build-world/tmp/work/i586-deby-linux/confuse/2.7-r0/git/support/config.guess (lines 7 through to 25) to verify that the LICENSE value "GPLv2+ & GPLv3+" remains valid [license-checksum]
ERROR: confuse-2.7-r0 do_populate_lic: Fatal QA errors found, failing task.
ERROR: confuse-2.7-r0 do_populate_lic: Function failed: populate_lic_qa_checksum
```

hieunv-tsdv, 2021-04-15:
@zuka0828 san,

The License and the files checksum are incorrect. I fixed it in 55e3ab7

zuka0828, 2021-04-18:
Fixed by #286, thanks a lot ! @hieunv-tsdv

closed 2021-04-18 by zuka0828