meta-debian/meta-debian #281: [morty] glusterfs: do_compile() failed

zuka0828, 2021-04-02:
Build conditions: 

* poky: e56be3cee517c5262486136dbd6d649b68c3a8b7
* meta-debian: 2d75fea1c314c10cc03cb29d42bdb98f3c926247
* Host: Debian 8 amd64 (Docker container)
* Build command: `bitbake -k world`

Extra settings in `local.conf`:

```
MACHINE = "qemuarm"
DISTRO_FEATURES += "pam x11 opengl"
BBFILES = "${BBFILES_CORE} ${BBFILES_DEBIAN}"

EXCLUDE_FROM_WORLD_pn-sudo-ldap = "1"
EXCLUDE_FROM_WORLD_pn-neon-gnutls = "1"
EXCLUDE_FROM_WORLD_pn-libsasl2-modules-gssapi-mit = "1"
EXCLUDE_FROM_WORLD_pn-libpam-heimdal = "1"
```

Error:

```
[...]
arm-deby-linux-gnueabi-libtool: link: arm-deby-linux-gnueabi-gcc -march=armv5e -marm --sysroot=/home/deby/poky/build-world/tmp/sysroots/qemuarm -Wall -O0 -O2 -pipe -g -feliminate-unused-debug-types -fdeb\
ug-prefix-map=/home/deby/poky/build-world/tmp/work/armv5e-deby-linux-gnueabi/glusterfs/3.5.2-r0=/usr/src/debug/glusterfs/3.5.2-r0 -fdebug-prefix-map=/home/deby/poky/build-world/tmp/sysroots/x86_64-linux=\
 -fdebug-prefix-map=/home/deby/poky/build-world/tmp/sysroots/qemuarm= -Wformat -Werror=format-security -Werror=implicit-function-declaration -I/home/deby/poky/build-world/tmp/sysroots/qemuarm/usr/include\
/python2.7 -I/home/deby/poky/build-world/tmp/sysroots/qemuarm/usr/include/python2.7 -fno-strict-aliasing -DNDEBUG -g -O3 -Wall -Wstrict-prototypes -O2 -pipe -g -feliminate-unused-debug-types -fdebug-pref\
ix-map=/home/deby/poky/build-world/tmp/work/armv5e-deby-linux-gnueabi/python/2.7.9-r1.3=/usr/src/debug/python/2.7.9-r1.3 -fdebug-prefix-map=/home/deby/poky/build-world/tmp/sysroots/x86_64-linux= -fdebug-\
prefix-map=/home/deby/poky/build-world/tmp/sysroots/qemuarm= -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/deby/poky/build-world/tmp/work/armv5e-deby-linux-gnueabi/glusterfs/3.5.2-\
r0=/usr/src/debug/glusterfs/3.5.2-r0 -fdebug-prefix-map=/home/deby/poky/build-world/tmp/sysroots/x86_64-linux= -fdebug-prefix-map=/home/deby/poky/build-world/tmp/sysroots/qemuarm= -Wformat -Werror=format\
-security -Werror=implicit-function-declaration -rdynamic -Wl,-O1 -Wl,--hash-style=gnu -Wl,-export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions -o .libs/gluster cli.o registry.o input.o cli-cmd.o cli-rl.o cl\
i-cmd-volume.o cli-cmd-peer.o cli-rpc-ops.o cli-cmd-parser.o cli-cmd-system.o cli-cmd-misc.o cli-xml-output.o cli-quotad-client.o  -L/home/deby/poky/build-world/tmp/sysroots/qemuarm/usr/lib/arm-linux-gnu\
eabi -L/home/deby/poky/build-world/tmp/sysroots/x86_64-linux/usr/lib ../../libglusterfs/src/.libs/libglusterfs.so -lreadline -lncurses ../../rpc/xdr/src/.libs/libgfxdr.so /home/deby/poky/build-world/tmp/\
work/armv5e-deby-linux-gnueabi/glusterfs/3.5.2-r0/build/rpc/rpc-lib/src/.libs/libgfrpc.so ../../rpc/rpc-lib/src/.libs/libgfrpc.so /home/deby/poky/build-world/tmp/work/armv5e-deby-linux-gnueabi/glusterfs/\
3.5.2-r0/build/libglusterfs/src/.libs/libglusterfs.so -lpython2.7 -lutil /home/deby/poky/build-world/tmp/sysroots/qemuarm/usr/lib/arm-linux-gnueabi/libxml2.so -lz -lm -ldl -lpthread -lcrypto
ERROR: oe_runmake failed
/home/deby/poky/build-world/tmp/sysroots/x86_64-linux/usr/lib/libncurses.so.5: file not recognized: File format not recognized
collect2: error: ld returned 1 exit status
Makefile:472: recipe for target 'gluster' failed
```

hieunv-tsdv, 2021-04-15:
@zuka0828 san,

```
/configure.ac:
LDFLAGS="-L`${PYTHON}-config --prefix`/lib `${PYTHON}-config --ldflags`"
```
"python-config --prefix" command always return native sysroot, it causes some error  when cross-compiling
edac8fa: Please check this commit and let me know if you have any comment.

zuka0828, 2021-04-18:
Fixed by #286, thanks a lot ! @hieunv-tsdv

closed 2021-04-18 by zuka0828