meta-debian/meta-debian #282: [morty] memtest86[+]: do_compile() failed

zuka0828, 2021-04-02:
Build conditions & Extra settings in local.conf: Same as #281 

Error (memtest86+):

```
| NOTE: make memtest.bin memtest CC=arm-deby-linux-gnueabi-gcc  -march=armv5e -marm --sysroot=/home/deby/poky/build-world/tmp/sysroots/qemuarm  LD=arm-deby-linux-gnueabi-ld --sysroot=/home/deby/poky/build-world/tmp/sysroots/qemuarm  -z max-page-size=0x1000 --hash-style=gnu AS=arm-deby-linux-gnueabi-as  -32
| ERROR: oe_runmake failed
| arm-deby-linux-gnueabi-gcc  -march=armv5e -marm --sysroot=/home/deby/poky/build-world/tmp/sysroots/qemuarm  -E -traditional head.S -o head.s
| arm-deby-linux-gnueabi-as  -32   -o head.o head.s
| arm-deby-linux-gnueabi-as: unrecognized option '-32'
| <builtin>: recipe for target 'head.o' failed
| make: *** [head.o] Error 1
```

Error(memtest86):

```
| NOTE: make CC=arm-deby-linux-gnueabi-gcc  -march=armv5e -marm --sysroot=/home/deby/poky/build-world/tmp/sysroots/qemuarm LD=arm-deby-linux-gnueabi-ld --sysroot=/home/deby/poky/build-world/tmp/sysroots/qemuarm  AS=arm-deby-linux-gnueabi-as  -32
| ERROR: oe_runmake failed
| rm -f *.o *.s *.iso memtest.bin memtest memtest_shared \
|       memtest_shared.bin memtest.iso
| arm-deby-linux-gnueabi-gcc  -march=armv5e -marm --sysroot=/home/deby/poky/build-world/tmp/sysroots/qemuarm -E -traditional head.S -o head.s
| arm-deby-linux-gnueabi-as  -32   -o head.o head.s
| arm-deby-linux-gnueabi-as: unrecognized option '-32'
| <builtin>: recipe for target 'head.o' failed
| make: *** [head.o] Error 1
| WARNING: exit code 1 from a shell command.
```

zuka0828, 2021-04-02:
memtest86[+] packages are only for x86 architectures.
Both recipes should limit the target architectures.

hieunv-tsdv, 2021-04-15:
@zuka0828 san,

> Both recipes should limit the target architectures.

4244574: in this commit, I added the COMPATIBLE_HOST variable to limit the target architectures (only for x86)

zuka0828, 2021-04-18:
Fixed by #286, thanks a lot ! @hieunv-tsdv

closed 2021-04-18 by zuka0828