meta-debian/meta-debian #284: [morty] u-boot-mkimage: do_compile() failed

zuka0828, 2021-04-02:
Build conditions & Extra settings in local.conf: Same as #281

Error:

```
[...]
cat ./Licenses/gpl-2.0.txt | gzip -9 -c | \
        tools/bin2header license_gzip > ./include/license.h
[...]
/lib/ld-linux.so.3: No such file or directory
[...]
tools/Makefile:242: recipe for target 'include/license.h' failed
make[1]: *** [include/license.h] Error 255
```

hieunv-tsdv, 2021-04-15:
@zuka0828 san,

> tools/bin2header license_gzip > ./include/license.h

The *bin2header* is generated by cross-compile, so it can't run on host machine.
8a8d207: this commit can be fix this issue, please help to check it.

zuka0828, 2021-04-18:
Fixed by #286, thanks a lot ! @hieunv-tsdv

closed 2021-04-18 by zuka0828