meta-debian/meta-debian #294: [morty] gccmakedep: do_package() fails

zuka0828, 2021-09-02:
I got the following error while building gccmakedep added by f03131a8f3273eb973446013442cf5b35debc36a.
The error does not occur in the second bitbake command.
Are there any missing build dependencies in gccmakedep?

Build conditions:

* metadata revisions
    * poky: e56be3cee517c5262486136dbd6d649b68c3a8b7
    * meta-debian: 7c6a6a80435df0b80541805fabde476acf913fc7
* `MACHINE`: `qemux86`
* Extra definitions in `local.conf`:
    * `IMAGE_INSTALL_append = " binutils libc6-dev libgcc libgcc-dev make minizip xutils-dev"`
    * `DISTRO_FEATURES += "x11"`
* Build command: `bitbake core-image-minimal`

Error:

```
ERROR: gccmakedep-7.7+3-r0.0 do_package: Error executing a python function in exec_python_func() autogenerated:

The stack trace of python calls that resulted in this exception/failure was:
File: 'exec_python_func() autogenerated', lineno: 2, function: <module>
     0001:
 *** 0002:fixup_perms(d)
     0003:
File: '/home/deby/poky/meta/classes/package.bbclass', lineno: 775, function: fixup_perms
     0771:    for path in target_path_vars:
     0772:        dir = d.getVar(path, True) or ""
     0773:        if dir == "":
     0774:            continue
 *** 0775:        fs_perms_table[dir] = fs_perms_entry(bb.data.expand("%s 0755 root root false - - -" % (dir), d))
     0776:
     0777:    # Now we actually load from the configuration files
     0778:    for conf in get_fs_perms_list(d).split():
     0779:        if os.path.exists(conf):
File: '/home/deby/poky/meta/classes/package.bbclass', lineno: 649, function: __init__
     0645:            lsplit = line.split()
     0646:            if len(lsplit) == 3 and lsplit[1].lower() == "link":
     0647:                self._setlink(lsplit[0], lsplit[2])
     0648:            elif len(lsplit) == 8:
 *** 0649:                self._setdir(lsplit[0], lsplit[1], lsplit[2], lsplit[3], lsplit[4], lsplit[5], lsplit[6], lsplit[7])
     0650:            else:
     0651:                msg = "Fixup Perms: invalid config line %s" % line
     0652:                package_qa_handle_error("perm-config", msg, d)
     0653:                self.path = None
File: '/home/deby/poky/meta/classes/package.bbclass', lineno: 661, function: _setdir
     0657:            self.path = os.path.normpath(path)
     0658:            self.link = None
     0659:            self.mode = self._procmode(mode)
     0660:            self.uid  = self._procuid(uid)
 *** 0661:            self.gid  = self._procgid(gid)
     0662:            self.walk = walk.lower()
     0663:            self.fmode = self._procmode(fmode)
     0664:            self.fuid = self._procuid(fuid)
     0665:            self.fgid = self._procgid(fgid)
File: '/home/deby/poky/meta/classes/package.bbclass', lineno: 692, function: _procgid
     0688:                return -1
     0689:            elif gid.isdigit():
     0690:                return int(gid)
     0691:            else:
 *** 0692:                return grp.getgrnam(gid).gr_gid
     0693:
     0694:        # Use for debugging the entries
     0695:        def __str__(self):
     0696:            if self.link:
Exception: KeyError: 'getgrnam(): name not found: root'

ERROR: gccmakedep-7.7+3-r0.0 do_package: Function failed: fixup_perms
ERROR: Logfile of failure stored in: /home/deby/poky/build-qemux86/tmp/work/i586-deby-linux/gccmakedep/7.7+3-r0.0/temp/log.do_package.1809225
ERROR: Task (/home/deby/poky/meta-debian/recipes-debian/xorg-util/gccmakedep_debian.bb:do_package) failed with exit code '1'
```

hieunv-tsdv, 2021-09-03:
Hi @zuka0828 san,

I created new build-tree and use same your build conditions, but I can't reproduce this issue.

Both of build commands (on docker):
 - bitbake core-image-minimal
 - bitbake gccmakedep (new build tree)

zuka0828, 2021-09-04:
Hi @hieunv-tsdv san,
Thank you for trying builds.
I'm also not able to reproduce the issue now, so I close this and reopen when it happens again.

closed 2021-09-04 by zuka0828