meta-debian/meta-debian #301: I am getting meta debian compile error..

er-rahul-km, 2022-03-08:
Hi,
I am using ubuntu 20.04 .
I am grtting compile errors.
I posted my logs here. Please provide solution to this.
 
$ bitbake core-image-minimal
WARNING: Host distribution "ubuntu-20.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Parsing recipes: 100% |#################################################################################################################################################| Time: 0:00:37
Parsing of 1041 .bb files complete (0 cached, 1041 parsed). 1821 targets, 234 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.42.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "arm-deby-linux-gnueabi"
MACHINE              = "qemuarm"
DISTRO               = "deby-tiny"
DISTRO_VERSION       = "10.0"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            = "warrior:d4b57c68b22027c2bedff335dee06af963e4f8a8"
meta-debian          = "warrior:4336f145b0aa649fcda1a28e762df0a4e5590e81"

NOTE: Fetching uninative binary shim from http://downloads.yoctoproject.org/releases/uninative/2.9/x86_64-nativesdk-libc.tar.xz;sha256sum=d07916b95c419c81541a19c8ef0ed8cbd78ae18437ff28a4c8a60ef40518e423
Initialising tasks: 100% |##############################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 282 Found 0 Missed 282 Current 0 (0% match, 0% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
WARNING: base-passwd-3.5.46-r0 do_fetch: Failed to fetch URL http://ftp.debian.org/debian/pool/main/b/base-passwd/base-passwd_3.5.46.tar.xz;name=base-passwd_3.5.46.tar.xz, attempting MIRRORS if available
ERROR: qemu-native-3.1-r0 do_compile: oe_runmake failed
ERROR: qemu-native-3.1-r0 do_compile: Function failed: do_compile (log file is located at /home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/temp/log.do_compile.894213)
ERROR: Logfile of failure stored in: /home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/temp/log.do_compile.894213
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 LD=ld  AR=ar OBJCOPY=objcopy LDFLAGS=-L/home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/recipe-sysroot-native/usr/lib                         -L/home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath,/home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/recipe-sysroot-native/lib                         -Wl,-O1 -Wl,--allow-shlib-undefined -Wl,--dynamic-linker=/home/rahul/MIRION_PROJECT/yocto_build/build/tmp/sysroots-uninative/x86_64-linux/lib/ld-linux-x86-64.so.2 -fuse-ld=bfd
|   GEN     aarch64-linux-user/config-devices.mak.tmp
|   GEN     i386-linux-user/config-devices.mak.tmp
|   GEN     arm-linux-user/config-devices.mak.tmp
|   GEN     mips-linux-user/config-devices.mak.tmp
|   GEN     mipsel-linux-user/config-devices.mak.tmp
|   GEN     mips64-linux-user/config-devices.mak.tmp
|   GEN     mips64el-linux-user/config-devices.mak.tmp
|   GEN     ppc-linux-user/config-devices.mak.tmp
|   GEN     aarch64-linux-user/config-devices.mak
|   GEN     i386-linux-user/config-devices.mak
|   GEN     mipsel-linux-user/config-devices.mak
|   GEN     arm-linux-user/config-devices.mak
|   GEN     riscv32-linux-user/config-devices.mak.tmp
|   GEN     mips-linux-user/config-devices.mak
|   GEN     riscv64-linux-user/config-devices.mak.tmp
|   GEN     mips64-linux-user/config-devices.mak
|   GEN     mips64el-linux-user/config-devices.mak
|   GEN     ppc-linux-user/config-devices.mak
|   GEN     sh4-linux-user/config-devices.mak.tmp
|   GEN     x86_64-linux-user/config-devices.mak.tmp
|   GEN     config-host.h
|   GEN     riscv32-linux-user/config-devices.mak
|   GEN     riscv64-linux-user/config-devices.mak
|   GEN     qemu-options.def
|   GEN     sh4-linux-user/config-devices.mak
|   GEN     qapi-gen
|   GEN     x86_64-linux-user/config-devices.mak
|   GEN     trace/generated-tcg-tracers.h
|   GEN     trace/generated-helpers-wrappers.h
|   GEN     trace/generated-helpers.h
|   GEN     trace/generated-helpers.c
|   GEN     module_block.h
|   GEN     ui/input-keymap-atset1-to-qcode.c
[... 955 lines cut ...]
|   CC      aarch64-linux-user/linux-user/linuxload.o
| make[1]: *** [/home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/qemu-3.1.0/rules.mak:69: linux-user/syscall.o] Error 1
| make[1]: *** Waiting for unfinished jobs....
| make: *** [Makefile:483: subdir-arm-linux-user] Error 2
| In file included from /usr/include/string.h:495,
|                  from /home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/qemu-3.1.0/include/qemu/osdep.h:84,
|                  from /home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/qemu-3.1.0/linux-user/elfload.c:2:
| In function ‘strncpy’,
|     inlined from ‘fill_psinfo’ at /home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/qemu-3.1.0/linux-user/elfload.c:3163:12,
|     inlined from ‘fill_note_info’ at /home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/qemu-3.1.0/linux-user/elfload.c:3345:5,
|     inlined from ‘elf_core_dump’ at /home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/qemu-3.1.0/linux-user/elfload.c:3494:9:
| /usr/include/x86_64-linux-gnu/bits/string_fortified.h:106:10: warning: ‘__builtin_strncpy’ specified bound 16 equals destination size [-Wstringop-truncation]
|   106 |   return __builtin___strncpy_chk (__dest, __src, __len, __bos (__dest));
|       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| make: *** [Makefile:483: subdir-aarch64-linux-user] Error 2
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/rahul/MIRION_PROJECT/yocto_build/build/tmp/work/x86_64-linux/qemu-native/3.1-r0/temp/log.do_compile.894213)
ERROR: Task (/home/rahul/MIRION_PROJECT/yocto_build/poky/meta-debian/recipes-debian/qemu/qemu-native_debian.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1153 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/rahul/MIRION_PROJECT/yocto_build/poky/meta-debian/recipes-debian/qemu/qemu-native_debian.bb:do_compile
Summary: There were 2 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

hiraku-wfs, 2022-03-08:
@er-rahul-km The following MR may be helpful to you.
https://github.com/miraclelinux/meta-emlinux/pull/118

zuka0828, 2022-03-09:
@er-rahul-km Thanks for reporting the problem.I believe that the patches in the link from @hiraku-wfs san solve the build problems with Ubuntu 20.04. meta-debian (warrior) itself does not support Ubuntu 20.04 as the build environment, but it's worth considering porting such advanced efforts.

@hiraku-wfs san, Thank you for your input. Let me check if porting the patches to meta-debian as well though it would take a bit time.

hiraku-wfs, 2022-03-11:
Hi, @er-rahul-km
> The patch is given for 3.0.0.

I think you are seeing the original patches of yocto, but those patches are rebased onto 3.1.0 in the PR.

hiraku-wfs, 2022-03-11:
@zuka0828 Thank you. If you need a PR from CTJ for meta-debian, please let me know.

hiraku-wfs, 2022-03-11:
@zuka0828 @er-rahul-km I'm sorry, there was a missing piece. The following patch is also necessary.

https://github.com/miraclelinux/meta-emlinux/blob/edb9ef70c43012aa3afa08d19604aa000b816737/recipes-debian/qemu/qemu/0014-linux-user-fix-to-handle-variably-sized-SIOCGSTAMP-w-custom.patch

zuka0828, 2022-09-02:
@er-rahul-km I confirmed that the problem of qemu-native on Ubuntu 20.04 is solved by the commit c1fd6102bed7ac8b5531437bf4bf694c57931863 which just applies patches developed by meta-emlinux project.

@hiraku-wfs Appreciate for sharing the information. I directly imported the four patches from your project to fix build issues on Ubuntu 20.04. They are applied to qemu-native (but not to qemu target because the all issues should be caused by host environment conditions).

closed 2022-09-02 by zuka0828

hiraku-wfs, 2022-09-05:
@zuka0828 Got it. Thanks!