meta-debian/meta-debian #311: docker: POKY_GIT_URL is not working anymore

jeanparpaillon, 2022-08-02:
Should be changed from https://git.yoctoproject.org/git/poky to https://git.yoctoproject.org/poky

jeanparpaillon, 2022-08-02:
My bad, error does not come from wqrong URL.
May it be the proxy ?

Anyway, I get the following error when building container from `docker` directory:

```
Cloning into '/home/deby/poky'...
fatal: unable to access 'https://git.yoctoproject.org/poky/': server certificate verification failed. CAfile: none CRLfile: none
ERROR: Service 'work' failed to build : The command '/bin/sh -c git clone $POKY_GIT_URL $HOME/poky' returned a non-zero code: 128
```

jeanparpaillon, 2022-08-02:
Adding `ca-certificates` package won't help

zuka0828, 2022-08-02:
Hi @jeanparpaillon , which branch are you using? If you are using `Dockerfile` in "morty" branch, please update the docker image using the latest `Dockerfile` which enables Extended LTS repository. Older packages in the container cause the error you are facing.

Reference: https://gitlab.com/cip-project/cip-core/deby/-/issues/15

zuka0828, 2022-09-02:
No update for around a month, so I'd like to close this issue.
Feel free to reopen if the issue has not been solved yet.

closed 2022-09-02 by zuka0828