meta-debian/meta-debian #326: Need suitable commit ID of meta-debian layer

manojrampur, 2023-05-16:
We are currently working on installing the .deb package into our target device by cloning the warrior branch of the meta-debian layer. However, we are encountering difficulties in identifying the appropriate commit ID that aligns with our Yocto build.

Here are the commit IDs we have:

meta-iotedge: db4541e7c9edb718682275ec1816ac2ed163e8a7
meta-rust: 5656b5ad104bbbbe1b711e3bbe35c88f0d9c9a05
poky: a307ef21c1acd999f67af0fc6cccffab7cc564b5

We need assistance in determining the suitable commit ID that will work effectively with our Yocto build.

zuka0828, 2023-12-09:
meta-debian is only compatible with the following OE-Core branch, which are already outdated.

* meta-debian morty branch : poky morty branch
* meta-debian warrior branch : poky warrior branch

I guess you are now trying to use meta-debian with quite fresh commits of poky and third-party layers.
They are completely incompatible and no sense to take efforts for.

meta-iotedge has ["warrior" branch](https://github.com/Azure/meta-iotedge/tree/warrior), and meta-rust has ["morty"](https://github.com/meta-rust/meta-rust/tree/morty).
Each could work with the corresponding branch (morty/warrior) of meta-debian, the both may have not been tested by anyone though.
If your goal is to use all layers together, no quick solution (i.e. common branch) unfortunately.

zuka0828, 2024-03-31:
I'd like to close this issue because quarter of the year passed. Feel free to reopen if needed.

closed 2024-03-31 by zuka0828