meta-debian/meta-debian #332: [warrior] The way to install /etc/hosts file, in order to resolve localhost in ptest

teradat, 2023-09-07:
# Issue
Ptests for some packages fail on deby(warrior) because they cannot resolve the localhost name.
Such packages I know:
- python3
- openssh

The ptest performed on meta-debian's Docker container is built with minimum configurations including the package to be tested.
But /etc/hosts file is not installed in the built image, so localhost name resolution cannot be performed.
As a result, the some networking-related tests fail.

## The way to reproduce with python3 package
On current warrior bitbake fails with openssl fetch.
So I used the #331 branch.

```
export TEST_PACKAGES="python3"
export TEST_DISTROS="deby"
export TEST_MACHINES="qemuarm64"
export COMPOSE_HTTP_TIMEOUT=7200
export PTEST_RUNNER_TIMEOUT=7200
$ cd ./docker
$ make qemu_ptest
```
### Result
../tests/logs/deby/qemuarm64/python3.ptest.log:
```
======================================================================
ERROR: test_HOST (test.test_support.TestSupport)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/usr/lib/python3.7/test/test_support.py", line 95, in test_HOST
    s.bind((support.HOST, 0))
socket.gaierror: [Errno -3] Temporary failure in name resolution
```
`support.HOST` is defined as "localhost".

```
test/support/__init__.py:628:HOST = "localhost"
```

# Investigation I did
## Summary
- In Debian 10, netbase uses postinst to create /etc/hosts.
- In Poky warrior, /etc/hosts creation(installation) is moved from netbase to base-files.
- In meta-debian warrior, neither of the above seems to be implemented

## Details
### meta-debian's netbase
  - a2678c22fda41bd84d1488f668a63ad6557b029e: Mar-28-2017, netbase: Install /etc/hosts and /etc/networks
  - 3f609957cd1c373ced7ad32d21e5d688f118eba0: Aug-16-2018, Remove unsupported recipes
  - cb53f139550ee0b405cba8bffa8633f5cfd36a05: Mar-28-2019, netbase: Add recipe based on Debian package (5.6)
    However, postinst was not implemented.

### meta-debian's base-files
There are no commits related to the /etc/hosts file.

### Poky's netbase
There is no processing related to the hosts file because it has already been moved to base-files.
Last commit of netbase recipe:
  - [09eda3d755def5a358bb500dfb6627a527dbb36d](https://git.yoctoproject.org/poky/commit/?h=warrior&id=09eda3d755def5a358bb500dfb6627a527dbb36d): Feb-17-2019, netbase: upgrade 5.5 -> 5.6

### Poky's base-files
  - [53cebfdfcc189a65de82eccdbd18acf4d979e624](https://git.yoctoproject.org/poky/commit/?h=warrior&id=53cebfdfcc189a65de82eccdbd18acf4d979e624): Jan-8-2019, netbase/base-files: move /etc/hosts from netbase to base-files
    This patch also fixes netbase_5.5.bb.
(Just FYI, emlinux is not affected by this issue because we use poky's base-files)

### Debian's netbase 5.6
Debian's netbase uses postinst to create /etc/hosts.
  - https://salsa.debian.org/md/netbase/-/blob/v5.6/debian/netbase.postinst

# Solution (idea)
I came up with two ideas below.
Currently, I think the better way is to modify the meta-debian's base-files recipe.
I would like to ask your opinion on how to solve this issue.

## Fixing netbase recipe
Add the `pkg_postinst_${PN}()` function to the recipe.
However, to run ptest of other package, we need to add netbase to RDEPENDS for the package.

## Fixing base-files recipe
Add hosts file to do_install of the recipe.
There are many differences in base-files recipes between Poky and meta-debian.
The base-files will be always installed, so we don't need to modify other packages.

zuka0828, 2023-12-09:
@teradat Thank you very much for pointing this issue out, and the investigation & suggestions.

In Debian, netbase is installed in the standard base system. Not installed only in the minimum base system (created with `debootstrap --variant=minbase`). So it's natural the system always relies on `/etc/hosts` provided by netbase.

For meta-debian, here is an aspect to decide solution:

* If the `localhost` definition in `/etc/hosts` is mandatory for the minimum system or not
  * If yes, base-files should provide it
  * If no, netbase should provide it, same as Debian

Based on this, netbase should provide it because

* It's not mandatory for the system where no networking even for localhost
* If the system needs localhost IP, it would need other basic TCP/IP functions from netbase as well

> However, to run ptest of other package, we need to add netbase to RDEPENDS for the package.

There is no issue about this because any packages should expressly have dependencies on requiring functions, of course, this needs additional efforts though.

teradat, 2023-12-12:
@zuka0828  
Thanks for your comment.
I will fix the netbase recipe.

teradat, 2024-03-13:
Fixed netbase package to install /etc/hosts and /etc/networks files.
See: https://github.com/meta-debian/meta-debian/pull/337

closed 2024-03-13 by teradat