meta-debian/meta-debian #333: support for my-mickledore

subrnath, 2023-10-14:
Does meta-debian master branch support poky my-mickledore release? Warrior release is quite old. When I am trying with poky my-mickledore, I am getting the following error -

source oe-init-build-env
Error: TEMPLATECONF value (which is /home/osboxes/debian/poky/meta-debian/conf) must point to meta-some-layer/conf/templates/template-name

zuka0828, 2023-12-09:
> Does meta-debian master branch support poky my-mickledore release?

No, it does not.
Sorry for confusing you. The master branch is originally the branch to develop recipes for poky master, but currently meta-debian has no plan to develop recipes for newer poky releases so the master branch is now just a clone of warrior branch. I deleted the master branch to avoid further confusions.

> Warrior release is quite old. When I am trying with poky my-mickledore, I am getting the following error -

Yes, the warrior branch (and master) is incompatible with poky mickledore (and any other non-warrior branches).
You need "mickledore" branch of meta-debian, but there is no plan to develop and maintain recipes for any new poky releases.

zuka0828, 2024-03-31:
I'd like to close this issue due to long enough passed

closed 2024-03-31 by zuka0828