meta-debian/meta-debian #336: Please bump vulnerable Expat to 2.5.0

hartwork, 2023-11-11:
Hi!

[libexpat 2.5.0](https://github.com/libexpat/libexpat/releases/tag/R_2_5_0) with **security fixes** has been released, the upstream [change log](https://github.com/libexpat/libexpat/blob/R_2_5_0/expat/Changes) has more details.

Thanks for updating! :pray: 

Best, Sebastian

zuka0828, 2023-12-09:
Thank you very much for the notification.

For not only the expat recipe but also all other recipes to build Debian source packages in meta-debian, there is no plan to upgrade the "upstream" version of Debian packages.
It's one of the important requirements for this project to keep functional compatibility for long-term.
Instead of upgrading upstream versions, recipes are getting security fixes from updates by Debian, Debian [LTS](https://www.freexian.com/lts/debian/), [ELTS](https://www.freexian.com/lts/extended/).

closed 2023-12-09 by zuka0828

hartwork, 2023-12-09:
@zuka0828 thanks for that info, I was not aware. I would like to note that [recipes-debian/sources/expat.inc@warrior](https://github.com/meta-debian/meta-debian/blob/warrior/recipes-debian/sources/expat.inc) is packaging libexpat `2.2.6-2+deb10u6` of bookworm which is *oldold*stable that is known to have [CVE-2013-0340](https://security-tracker.debian.org/tracker/CVE-2013-0340) unfixed. Given how much time I personally invested in fixing that vulnerability that's a bit of a pity and it makes meta-debian trivial to attack via a [billion laughs (denial of service) attack](https://en.wikipedia.org/wiki/Billion_laughs_attack) in 2023 while I [fixed it in 2021](https://blog.hartwork.org/posts/cve-2013-0340-billion-laughs-fixed-in-expat-2-4-0/) already. I guess that means that Expat in meta-debian will remain vulnerable.

zuka0828, 2024-01-09:
@hartwork Thank you for sharing the details.

> I would like to note that [recipes-debian/sources/expat.inc@warrior](https://github.com/meta-debian/meta-debian/blob/warrior/recipes-debian/sources/expat.inc) is packaging libexpat 2.2.6-2+deb10u6 of buster which is oldoldstable that is known to have [CVE-2013-0340](https://security-tracker.debian.org/tracker/CVE-2013-0340) unfixed.
> [...]
> I guess that means that Expat in meta-debian will remain vulnerable.

I think that's true.
First of all, appreciate your great works so far to fix the security issue.
In this situation, first we should communicate with Debian buster LTS (which will be ELTS from July this year [1]) to check the possibility of backporting the security fix.

However, according to the [expat changelog](https://github.com/libexpat/libexpat/blob/R_2_4_1/expat/Changes#L19-L49), the fix requires multiple API changes which would not be acceptable in long-term maintenance such as Debian LTS/ELTS.
I guess the [PR466](https://github.com/libexpat/libexpat/pull/466) (+ PR484) is the main changes of the security fix. This consists of a big code changes (additions) especially for `lib/xmlparse.c` that I'm feeling the backport is quite hard.
Anyway let me try to share this information with LTS.

[1] https://wiki.debian.org/LTS/Extended

hartwork, 2024-01-09:
> However, according to the [expat changelog](https://github.com/libexpat/libexpat/blob/R_2_4_1/expat/Changes#L19-L49), the fix requires multiple API changes which would not be acceptable in long-term maintenance such as Debian LTS/ELTS.

Additions, not API-breaking changes, but yes. It's quite a bit, yes.

> I guess the [PR466](https://github.com/libexpat/libexpat/pull/466) (+ PR484) is the main changes of the security fix. This consists of a big code changes (additions) especially for `lib/xmlparse.c` [...]

I confirm that it's these two pull requests:
- https://github.com/libexpat/libexpat/pull/484
- https://github.com/libexpat/libexpat/pull/484

My personal recommendation is to bump Expat to 2.5.0 everywhere rather than backporting pieces. It will not be the last hard-to-backport security fix with Expat, backporting will either not be economic or not secure (when incomplete) with Expat. If making Expat an exception like that (i.e. latest releases instead of expensive risky backports), please take my vote for it.